SpacehuhnTech/esp8266_deauther #973: OLED SSIDs menu not working

OstenTV, 2018-10-29:
I have a DSTRIKE v3.5 (the one with RGB LED), running Deauther v2.0.5.
The menu for selecting SSIDs doesn't open on the OLED interface. If I enable random mode from the WEB interface, I can enter the menu without any problem.

elliot-holley, 2018-11-07:
Same problem here

elliot-holley, 2018-11-07:
I have a feeling its something to do with something has to be actually in the menu to open it
just a guess

elliot-holley, 2018-11-07:
I have a solve!

Go to web interface, type in whatever you like in 'SSID', In the 'Number' selection type '24' (or whatever amount you like) press save and then reboot! for some reason this does not get deleted when the board is rebooted

hopes this solves your problem!

tobozo, 2018-11-07:
reminder : ssid must be between 8 ans 31 bytes (some unicode letters can use up to 3 bytes) otherwise WiFi won't work as intended

so you can't just type in *whatever you like* in 'SSID'

elliot-holley, 2018-11-07:
oh

OstenTV, 2018-11-07:
No.... Lemme explain. The menu won't open when the list is empty. If I go to the WEB interface and add an SSID, it works fine. It has nothing to do with the SSID being too long, as there are none.

elliot-holley, 2018-11-08:
this is a feature?

elliot-holley, 2018-11-08:
The problem Isn't really not being able to access the list when there are no SSIDs; the problem is that there dosen't appear to be any way to add a SSID to the list from the display/buttons alone.

OstenTV, 2018-11-08:
You can add SSIDs from inside the SSIDs menu (at the bottom). But if you can't enter the menu, then there would be no way to add them, so you're kinda right. In the previous version, you were able to enter the menu while the list was empty.

elliot-holley, 2018-11-08:
[image]
Remember the good ol' days?

elliot-holley, 2018-11-08:
Don't we all?

Anyway, back on to the main subject of this thread. An easy fix would be to go on to the web interface, just put a space in the SSID name, put the number of SSIDs as 1 and save. in my experience this does not get deleted when I reboot mine. I know it's not permanent, but it should work for the moment, until a proper fix is rolled out by someone with more experience than me in coding.

elliot-holley, 2018-11-08:
I know that sounds lazy, but I have no experience with coding, apart from python.

spacehuhn, 2018-11-08:
If you compile the latest version yourself, it should work. 
Might not have been in version 2.0.5. Release of v2.1 is coming soon, so don't worry ;)

elliot-holley, 2018-11-09:
:)

closed 2019-10-09 by tobozo