SpacehuhnTech/esp8266_deauther #975: Category '' in library ArduinoJson is not valid. Setting to 'Uncategorized'

nc9710, 2018-11-02:
Having an error when compiling, ArduinoJSON is v5. Please help.

WARNING: Category '' in library ArduinoJson is not valid. Setting to 'Uncategorized'
esp8266_deauther:21:2: error: #error Please upgrade/downgrade ArduinoJSON library to version 5!

 #error Please upgrade/downgrade ArduinoJSON library to version 5!

  ^

esp8266_deauther:24:17: error: oui.h: No such file or directory

 #include "oui.h"

                 ^

compilation terminated.

exit status 1
#error Please upgrade/downgrade ArduinoJSON library to version 5!

mrohit2011, 2018-11-10:
I am too facing the same issue.

forseenink3238, 2018-11-29:
Me too but the source I actually having a problem

MartieTheSmartie, 2018-12-13:
Same problem @spacehuhn is this fixable?

spacehuhn, 2018-12-14:
The obvious fix is to comment out the 3 lines (17,21,22) that are causing the error: 
https://github.com/spacehuhn/esp8266_deauther/blob/master/esp8266_deauther/esp8266_deauther.ino#L17

But you still have to install the correct ArduinoJSON library version!
Check the Arduino library manager, your Arduino library folder and select the right board (under deauther not esp8266 section).

spacehuhn, 2019-01-04:
Check your Arduino library folder (C:\Users\user\Documents\Arduino\libraries) and delete the ArduinoJSON folder if it exists. 
If the library in there is the wrong version, you might not be able to up/downgrade it from the Arduino library manager, so you have to delete it maybe.
The problem is that Arduino gives priority to the libraries in your Documents\Arduino folder instead of the one that comes with the deauther pboard ackage.

Pro Tip: Read the error message carefully! It usually tells you the error 😉

Cubensiss, 2019-01-06:
In this folder is not ArduinoJSON folder only readmme datei

omega999, 2019-01-09:
esp8266_deauther:33:2: error: #error Please upgrade/downgrade ArduinoJSON library to version 5!

 #error Please upgrade/downgrade ArduinoJSON library to version 5!

  ^

esp8266_deauther:34:2: error: #endif without #if

 #endif

  ^

In file included from sketch\Settings.h:11:0,

                 from D:\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino:39:

A_config.h:7:24: error: SH1106Wire.h: No such file or directory

 #include "SH1106Wire.h"

                        ^

compilation terminated.

exit status 1
#error Please upgrade/downgrade ArduinoJSON library to version 5!

WARNING: Category '' in library ArduinoJson is not valid. Setting to 'Uncategorized'
esp8266_deauther:33:2: error: #error Please upgrade/downgrade ArduinoJSON library to version 5!

 #error Please upgrade/downgrade ArduinoJSON library to version 5!

  ^

esp8266_deauther:34:2: error: #endif without #if

 #endif

  ^

In file included from sketch\Settings.h:11:0,

                 from D:\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino:39:

A_config.h:7:24: error: SH1106Wire.h: No such file or directory

 #include "SH1106Wire.h"

                        ^

compilation terminated.

exit status 1
#error Please upgrade/downgrade ArduinoJSON library to version 5!

please help

tobozo, 2019-01-09:
I think the error message is trying to tell you to do this : 

[image]

Blare1337, 2019-01-17:
Im getting this error now with Arduino

Sketch uses 704766 bytes (92%) of program storage space. Maximum is 761840 bytes.
Global variables use 42772 bytes (52%) of dynamic memory, leaving 39148 bytes for local variables. Maximum is 81920 bytes.
Serial port not selected.

FaroJoaoFaro, 2019-01-23:
Hello
When I go to the Arduino Library I don't have ArduinoJson
What can I do?

sarimbinwaseem, 2019-01-29:
> Check your Arduino library folder (C:\Users\user\Documents\Arduino\libraries) and delete the ArduinoJSON folder if it exists.
> If the library in there is the wrong version, you might not be able to [...]

Still not working..... 
What I have done...
1. Deleted the JSON folder from Documents/Arduino
2. Downloaded Arduino JSON V 5.0.0 from page:  https://github.com/bblanchon/ArduinoJson/releases?after=v5.1.0-beta.1
3. Included in IDE.
4. Selected Node MCU 12E from Deauther Module section
But still problem is there..

sarimbinwaseem, 2019-01-29:
> Hello
> When I go to the Arduino Library I don't have ArduinoJson
> What can I do?

Download it from https://github.com/bblanchon/ArduinoJson/releases?after=v5.1.0-beta.1

labomb, 2019-02-11:
In my case, I've determined that the problem was that the deauther board package that is required to be installed via the board manager (http://phpsecu.re/esp8266/package_deauther_index.json) includes the ArduinoJson library. On my Windows system, it was found here after installation:

%USERPROFILE%\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries

I deleted the ArduinoJson directory located there, and then installed the latest version 5 iteration (in my case, version 5.13.4) via the IDE Library Manager . This will install the ArduinoJson lib in your Documents folder (specifically, %USERPROFILE%\Documents\Arduino\libraries).

Oddly, the ArduinoJson library included with the deauther board package appears to be a valid version (5.13.0)... but I still got the upgrade/downgrade error message until it was deleted. Haven't had an issue since.

jumblies, 2019-02-14:
Confirmed working on PRE v6 releases.  Compiled today

tobozo, 2019-02-15:
@labomb @jumblies thanks for reporting this, the deauther SDK is now [bundled with ArduinoJSON 5.13.14](https://github.com/tobozo/Arduino/blob/deauther/libraries/ArduinoJson/library.properties)

you will need to remove/reinstall it from the boards manager to see the effects though 

can you please confirm this action fixed the issue ?

jumblies, 2019-02-15:
I can say for certain that ArduinoJSON 5.13.14 will allow compilation; will try to test removing and reinstalling later

But then I ran into the SPIFFS format bootloop.  Reverting to an earlier version got me a working device.  I'll see which one I'm using and post back later.

labomb, 2019-02-15:
@tobozo that took care of the problem. I removed the locally installed ArduinoJson lib and removed/reinstalled the deauth board package as requested and it appears that the updated ArduinoJson library that is now bundled works fine... no version errors during build. Thanks for the update.

nc9710, 2019-03-22:
Followed advice by spacehuhn and resolved issue. I also deleted previously installed libraries which got rid of some warnings I had while compiling. Also, a quick work around (if engaging in the deauther project) is to download the latest bin file and use the ESP8266Flasher.

tytower, 2019-08-18:
Importantly for those installing on Arduino , now until they change to ver 6  you can install the Arduinojson version 5 but the error message then will tell you you have 2  Arduinojson libraries and importantly will give you the address of the version 6 it tried to use . Note that directory address down ,go there and delete the whole file first then restart.

Scooby2001, 2019-11-16:
Hello there,
i checked every setting. Downgraded ArduinoJson library to version 5.13.4.
I tried everything that was mansioned here but still the same problem:

Arduino: 1.8.10 (Windows 7), Board: "Generic ESP8266 Module, Serial, 160 MHz, 80MHz, DIO, 115200, 4M (3M SPIFFS), nodemcu"

Mehrere Bibliotheken wurden für "EEPROM.h" gefunden
 Benutzt: C:\Users\Abwickler\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\EEPROM
Mehrere Bibliotheken wurden für "ArduinoJson.h" gefunden
 Benutzt: C:\Users\Abwickler\Documents\Arduino\libraries\ArduinoJson
 Nicht benutzt: C:\Users\Abwickler\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\ArduinoJson
esp8266_deauther:24:17: error: oui.h: No such file or directory

 #include "oui.h"

                 ^

compilation terminated.

exit status 1
oui.h: No such file or directory

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.

tobozo, 2019-11-16:
you still have two versions, you can safely delete the first folder (Benutzt)

    Benutzt: C:\Users\Abwickler\Documents\Arduino\libraries\ArduinoJson

    Nicht benutzt: C:\Users\Abwickler\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\ArduinoJson

Scooby2001, 2019-11-16:
Thanks for youre help, tried it but still the same problem... 

Arduino: 1.8.10 (Windows 7), Board: "Generic ESP8266 Module, Serial, 160 MHz, 80MHz, DOUT, 115200, 4M (3M SPIFFS), nodemcu"

esp8266_deauther:23:17: error: oui.h: No such file or directory

 #include "oui.h"

                 ^

compilation terminated.

Mehrere Bibliotheken wurden für "ArduinoJson.h" gefunden
 Benutzt: C:\Users\Abwickler\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\ArduinoJson
Mehrere Bibliotheken wurden für "EEPROM.h" gefunden
 Benutzt: C:\Users\Abwickler\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\EEPROM
exit status 1
oui.h: No such file or directory

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.

Jacob12138xieyuan, 2019-11-29:
comment 17,21,22 line, install 5.x, works for me

anonymous7cyber, 2020-03-17:
> comment 17,21,22 line, install 5.x, works for me

Thanks man!

James0329, 2020-08-25:
I got a strange problem. friends.

If I don't delete checker all things will stop at version checker.
If I delete the version checker method.
Then install any 5.x ArduinoJson when I upload file to esp8266 will trigger a error like
"invalid initialization of non-const reference of type 'ArduinoJson::JsonArray&' from an rvalue of type 'ArduinoJson::JsonArray'
"
or
"no matching function for call to 'ArduinoJson::DynamicJsonBuffer::DynamicJsonBuffer(int)'
"
this problem location at ssid.cpp file.
"DynamicJsonBuffer jsonBuffer(4000);"
 "JsonArray & arr = obj.get<JsonArray>(str(SS_JSON_SSIDS)); "
I try 5.0.0|5.0.5|5.1.0|5.4.0 but nothing change.
Need some help  (T_T)

5ch4um1, 2021-01-03:
had this issue yesterday, but somehow it went away, i tried a few versions of ArduinoJson, to no avail, but today it works, not sure if it was the logout (restart in my case) ? I also deleted the repository and cloned it again, but i guess this shouldn't make a difference?