SpacehuhnTech/esp8266_deauther #977: Is it possible to target specific devices on a network instead of them all?

RottenDirector, 2018-11-06:
Title says it all

spacehuhn, 2018-11-06:
yes

RottenDirector, 2018-11-06:
> yes

How?

killergeek, 2018-11-06:
[image]
https://github.com/spacehuhn/esp8266_deauther/wiki

spacehuhn, 2018-11-07:
You scan for stations and select one, or add a device to the name list.

jLynx, 2018-11-07:
Yes.

guoqiyun, 2018-11-24:
It seems this was removed from recent versions for some reason. I flashed my board to test it on my mobile. I obviously would not want to target an entire network!
[image]

spacehuhn, 2018-11-24:
Scan for stations and select one, or add a device to the name list.

forseenink3238, 2018-11-29:
Just capture a WiFi handshake then you will get mac address. Use the mac address to add a device.

spacehuhn, 2018-11-29:
No need to capture any handshakes... just scan for stations.

closed 2018-11-29 by spacehuhn