SpacehuhnTech/esp8266_deauther #980: PUYA flash chip does not have SPIFFS available.

renat2985, 2018-11-12:
It appears that the PUYA flash chip is being adopted by a growing number of manufacturers of the esp8266 modules. My most recent purchases of esp-12 modules from 3 separate suppliers have all had the PUYA chip on-board (for a total of 30 modules). The issue is that Arduino IDE x.xx sketches using any 8266 core can not currently write to SPIFFS on modules with this chip.

Any ideas?

tobozo, 2019-10-09:
Closing this issue as inactive, feel free to reopen if necessary.

closed 2019-10-09 by tobozo

adapso, 2020-02-27:
I have the following problem, I was waiting for a correction for the PUYA problem for over a year.  This week after resuming the project I upgraded to the core version 2.6.3 of ESP826, for my happiness ESP returned to save the information correctly in the flash using SPIFFS, but to my sadness this version caused some problem with the ESP connection, I fired  a notification using the Pushover API.  In core version 2.4.2 the shooting works very well, but in version 2.6.3 it is not working.  Can anyone help?