SpacehuhnTech/esp8266_deauther #981: Is this operating system specific?

Moondog360, 2018-11-13:
I'm curios if this only affects some OS and not all, I have noticed that most mobile phones disconnect. Some hope onto the 5ghz which is normal, but when it comes to PC and Laptops I noticed that Ubuntu OS doesn't get affected even tho it's the same laptop that uses dual boot. Not sure what's happening. I asked few friends to test on their devices that have dual boot and the same thing happens. Windows OS disconnects but Ubuntu/Debian/Linux/Unix/Kali does not get affected by it. Any help please?

tobozo, 2018-11-13:
Some systems are faster to recover after a deauth than others, but some architectures are more resilient too.

I've seen an old iPhone starting to play a youtube video while being deauth'd a couple of times per second, and it was apparently unaffected by the disconnections.

Moondog360, 2018-11-13:
@tobozo interesting. It's weird because I don't even see it deauth at all. not even for a second. That's some fast recovery xD

vuong244, 2018-11-22:
Please attack the client, instead of attacking the access point. Everything will work fine

closed 2018-11-22 by Moondog360