SpacehuhnTech/esp8266_deauther #984: No pwned network

Johnsjs2, 2018-11-15:
So I flashed my Cp2102 with the 1mb flash .bin file with the NodeMCU program I looked at the logs on NodeMCU it says all program flashes were successful however it does not show a pwned network. I have tried 3 different cords, USB 3.0, USB 2.0, and a wall charger. On my phone, I tried connecting to a hidden network the ssid I put pwned and the security as WPA2 with the password as deauther. From what I can tell the flash was successful but it is not giving off the pwned network, any suggestions?

"""
Note:Detect serial port changed.
Note:Auto MAP serial port.Port-->COM3

Note:Serial port connected.
Note:Begin find ESP8266.
Note:ESP8266 ACK success.
Note:ESP8266 ACK success.
Note:Set base address success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
[... 1169 lines cut ...]
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Program flash success.
Note:Serial port disconnected.
Note:Detect serial port changed.
Note:Auto MAP serial port.Port-->COM1
""""

PrinceShaji, 2018-11-23:
If that doesn't work, try compiling the source.

sro2, 2018-11-26:
> Read the Wiki

I used node mcu flasher not adrino . So i flash very easily.

shindury26, 2018-12-10:
try to use serial monitor then type format and wait few sec try reset and wait again
if still not working try to use Reset_Sketch flash it and flash again the deauther 

sometimes its ruin if you the to on the hidden or tried different program they store settings
i remember few many  times i flash same firmware but when i reset restore to my first modified setting

RChadwick7, 2018-12-13:
I have this same issue. While I have successfully flashed to various devices with and without an OLED, I have two USB-Only devices that look like this: https://www.ebay.com/itm/113314595584?ViewItem=&item=113314595584
These don't work with the latest firmware no matter what I do. The SSID showed up as Fairy something. I've used the three flashers mentioned in the Wiki, I've used the 1mb, 4mb, and the USB version, with ethe same results.  I've gone down all the way to version 1.2. Version 1.2 actually worked. Going back to the latest firmware in the same device creates no SSID. Is there a known incompatibility between this device and the latest firmwares?

lukethomas01, 2019-01-05:
> I have this same issue. While I have successfully flashed to various devices with and without an OLED, I have two USB-Only devices that look like this: [...]

I've got a similar issue, I get the below wireless network. I've tested various settings but come up short everytime.

SSID:	ESP_265BEC
Protocol:	802.11g
Security type:	Open
Network band:	2.4 GHz
Network channel:	1
IPv4 address:	154.153.153.53
IPv4 DNS servers:	154.153.153.153
Manufacturer:	Qualcomm Atheros Communications Inc.
Description:	Qualcomm QCA61x4A 802.11ac Wireless Adapter
Driver version:	12.0.0.697
Physical address (MAC):	B0-52-16-31-69-37

Here the log from the flashing.

test offset :  507904 0x7c000
case ok
test check all res:   (True, [[u'C:\\Users\\xxx\\Downloads\\esp8266_deauther_1.0_1mb.bin', 507904]])
(True, [[u'C:\\Users\\xxx\\Downloads\\esp8266_deauther_1.0_1mb.bin', 507904]])
test running : False
serial port opened
Connecting...
chip sync ok!
chip_flg 1
r1: 26; r2:5b ; r3: ec
MAC AP : 1A-FE-34-26-XX-XX
MAC STA: 18-FE-34-26-XX-XX
filename:  C:\Users\xxx\Downloads\FLASH_DOWNLOAD_TOOLS_v0.9.6_150419\FLASH_DOWNLOAD_TOOLS_v0.9.6_150419\bin_tmp\esp8266_deauther_1.0_1mb.bin
offset :  507904
Erasing flash...
head:  4 ;total: 134
erase size :  532480
Writing at 0x00101000... (99 %)
Leaving...
com closed

closed 2019-10-09 by tobozo