SpacehuhnTech/esp8266_deauther #988: No AP shown, NodeMCU, error in log

spreadzz, 2018-11-22:
Mounting SPIFFS...OK
Formatting SPIFFS...OK
Switched to Channel 9
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40253ca9 depc=0x00000000

ctx: cont 
sp: 3fff1a00 end: 3fff1c50 offset: 01a0

>>>stack>>>
3fff1ba0:  3fff4a98 0000000f 0000000b 3fff4a98  
3fff1bb0:  40253ca9 00000016 00000006 3fff4f90  
3fff1bc0:  00000006 3fff4f90 3fff0780 402093db  
3fff1bd0:  3fff4ab8 0000000f 00000001 3fff2aa0  
3fff1be0:  3fff0920 3fff4cfc 00000000 00000000  
3fff1bf0:  00000000 3fff0944 3fff0b60 40267494  
3fff1c00:  3fff1d44 3fff0b60 3fff0944 4021f82b  
3fff1c10:  00000000 00000000 00000000 00000000  
3fff1c20:  00000000 00000000 00000000 3fff0c24  
3fff1c30:  3fffdc20 00000000 3fff0c1d 4022614e  
3fff1c40:  00000000 00000000 3fff0c30 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Mounting SPIFFS...OK
Formatting SPIFFS...OK
Switched to Channel 9
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40253ca9 depc=0x00000000

ctx: cont 
sp: 3fff1a00 end: 3fff1c50 offset: 01a0

>>>stack>>>
3fff1ba0:  3fff4a98 0000000f 0000000b 3fff4a98  
3fff1bb0:  40253ca9 00000016 00000006 3fff4f90  
3fff1bc0:  00000006 3fff4f90 3fff0780 402093db  
3fff1bd0:  3fff4ab8 0000000f 00000001 3fff2aa0  
3fff1be0:  3fff0920 3fff4cfc 00000000 00000000  
3fff1bf0:  00000000 3fff0944 3fff0b60 40267494  
3fff1c00:  3fff1d44 3fff0b60 3fff0944 4021f82b  
3fff1c10:  00000000 00000000 00000000 00000000  
3fff1c20:  00000000 00000000 00000000 3fff0c24  
3fff1c30:  3fffdc20 00000000 3fff0c1d 4022614e  
3fff1c40:  00000000 00000000 3fff0c30 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Mounting SPIFFS...OK
Formatting SPIFFS...

forseenink3238, 2018-11-29:
Did you flash it with a bin file or uploaded the code from arduino IDE

spreadzz, 2018-11-29:
I flashed it with a bin from here, then uploaded the code with arduino IDE.

I figured solved it eventually. For some reason it only works if I flash the bin after I upload the arduino code. Every time I upload a new arduino code on it, I have to flash it right after.

Also after reading all the documentation here, I noticed I was using a "fake" board.
It works now, but has a very low wifi signal.

Since then I purchased original Dstrike boards, I only tested one of them but works much better.

forseenink3238, 2018-11-29:
Ok good.

closed 2019-10-09 by tobozo