SpacehuhnTech/esp8266_deauther #995: Connection dorps on Wifi deather

don1406, 2018-11-28:
Hi,

I have recently bought a WIFI deauther (USB version), when i connect to it, it is going ok.
I ping the deauther, to see if it is going down or not.
I want to scan a network, or change from the tab "Scan" to "SSID" for example, then i get no ping and only request time out in my terminal, and the deauther doenst respond anymore. 

I have already flashed the deauther 2 times, but still same problem. i used verison 2.0.5

Does anyone know if it is the board or something ?

forseenink3238, 2018-11-28:
It was happening to me yesterday and my version was 2.0.5 so I flashed it with the new version 2.1.0 before posting an issue

don1406, 2018-11-28:
If you suggesting that i have to wait a year or so, no thats not an option. Maybe i can try flash it with version 2.1.0. Where did you get the 2.1.0 release ?

forseenink3238, 2018-11-28:
No not that

forseenink3238, 2018-11-28:
Should i send iy

don1406, 2018-11-28:
Hi, 
Thank you for being so helpfull. I have flashed the board with version 2.10.
Hoever, i get allot of error: cant ....script.json errors.. 
Am i the only one ?

forseenink3238, 2018-11-28:
No it happens to me too

forseenink3238, 2018-11-28:
can you help me in my issue please?

don1406, 2018-11-29:
Wich issue ?

closed 2018-11-29 by don1406