SpacehuhnTech/esp8266_deauther #1001: SimpleButton.h: No such file or directory

DavidRaid, 2018-12-06:
I have my board set to Generic ESP8266 Module which is under ESP8266 Deauther Modules.
Error log :
```
Build options changed, rebuilding all
In file included from sketch\CLI.h:18:0,

                 from E:\Downloads\esp8266_deauther-master\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino:32:

sketch\DisplayUI.h:12:26: fatal error: SimpleButton.h: No such file or directory

 #include <SimpleButton.h>

                          ^

compilation terminated.

exit status 1
Error compiling for board NodeMCU (ESP-12 Module).
```
Thanks in advance.

CodeNameHawk, 2018-12-09:
Try [https://github.com/spacehuhn/SimpleButton](url)
other libs
[https://github.com/adafruit/Adafruit_NeoPixel](url)
[https://github.com/ThingPulse/esp8266-oled-ssd1306](url)

spacehuhn, 2018-12-09:
Please follow the instructions from the wiki: https://github.com/spacehuhn/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide

Let us know if you get stuck somewhere.

deuxzero, 2018-12-12:
hello, i follow every step and also download the missing lib (oled,neopixel and simple) but now  i have got a new error when i try to compil 

`Arduino : 1.8.7 (Windows 10), Carte : "WeMos D1, 80 MHz, Serial, 115200, 4M (1M SPIFFS)"

sketch\Scan.cpp: In member function 'void Scan::start(uint8_t, uint32_t, uint8_t, uint32_t, bool, uint8_t)':

Scan.cpp:81:36: error: no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool)'

         WiFi.scanNetworks(true,true);

                                    ^

sketch\Scan.cpp:81:36: note: candidate is:

In file included from sketch\Accesspoints.h:5:0,

                 from sketch\Scan.h:5,

                 from sketch\Scan.cpp:1:

C:\Users\creaz\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:243:12: note: int8_t ESP8266WiFiClass::scanNetworks(bool)

     int8_t scanNetworks(bool async = false);

            ^

C:\Users\creaz\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:243:12: note:   candidate expects 1 argument, 2 provided

exit status 1
no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool)'

Ce rapport pourrait être plus détaillé avec
l'option "Afficher les résultats détaillés de la compilation"
activée dans Fichier -> Préférences.

Do you have any clue why?
Tk

tobozo, 2018-12-12:
oui

[image]

spacehuhn, 2018-12-12:
A tale of thousand and one issues :D

nityoday, 2019-01-12:
I dont undertand anything after I get the Simple  button error .
I was following this article :  https://www.instructables.com/id/DIY-Wifi-Jammer-With-ESP8266-and-Mobile-App/

error is 
" Arduino: 1.8.8 (Windows 10), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

Build options changed, rebuilding all
In file included from sketch\CLI.h:18:0,

                 from C:\Users\admin\Desktop\esp8266_deauther-master\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino:32:

DisplayUI.h:12:26: error: SimpleButton.h: No such file or directory

 #include <SimpleButton.h>

                          ^

compilation terminated.

exit status 1
SimpleButton.h: No such file or directory

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
"

Now what should I do please tell me?

tobozo, 2019-01-12:
[read the wiki](https://github.com/spacehuhn/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide)

closed 2019-10-09 by tobozo