SpacehuhnTech/esp8266_deauther #1002: I keep getting an error when i try to upload to my ESP8266. (PLEASE HELP)

xwhd12, 2018-12-06:
so it says this 

Arduino: 1.8.8 (Windows 10), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

In file included from sketch\CLI.h:18:0,

                 from sketch\CLI.cpp:1:

DisplayUI.h:71:9: error: 'Button' does not name a type

         Button* up   = NULL;

         ^

DisplayUI.h:72:9: error: 'Button' does not name a type

         Button* down = NULL;

         ^

In file included from sketch\CLI.h:11:0,

                 from sketch\CLI.cpp:1:

A_config.h:66:26: error: 'SSD1306Wire' does not name a type

 #define DEAUTHER_DISPLAY SSD1306Wire display = SSD1306Wire(0x3c, 5, 4); // for 0.96" OLED

                          ^

sketch\DisplayUI.h:76:9: note: in expansion of macro 'DEAUTHER_DISPLAY'

         DEAUTHER_DISPLAY // see config.h

         ^

sketch\Settings.cpp: In member function 'void Settings::load()':

Settings.cpp:17:15: error: 'data' was not declared in this scope

     version = data.get<String>("version");

               ^

Settings.cpp:17:30: error: expected primary-expression before '>' token

     version = data.get<String>("version");

                              ^

Settings.cpp:20:67: error: expected primary-expression before '>' token

     if (data.containsKey(keyword(S_SSID))) setSSID(data.get<String>(keyword(S_SSID)));

                                                                   ^

Settings.cpp:21:75: error: expected primary-expression before '>' token

     if (data.containsKey(keyword(S_PASSWORD))) setPassword(data.get<String>(keyword(S_PASSWORD)));

                                                                           ^

Settings.cpp:22:74: error: expected primary-expression before '>' token

     if (data.containsKey(keyword(S_CHANNEL))) setChannel(data.get<uint8_t>(keyword(S_CHANNEL)));

                                                                          ^

In file included from C:\Users\Leo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/Arduino.h:29:0,

[... 505 lines cut ...]
         Button* down = NULL;

         ^

In file included from sketch\Settings.h:11:0,

                 from C:\Users\Leo\Desktop\esp8266_deauther-2.1.0\esp8266_deauther\esp8266_deauther.ino:27:

A_config.h:66:26: error: 'SSD1306Wire' does not name a type

 #define DEAUTHER_DISPLAY SSD1306Wire display = SSD1306Wire(0x3c, 5, 4); // for 0.96" OLED

                          ^

sketch\DisplayUI.h:76:9: note: in expansion of macro 'DEAUTHER_DISPLAY'

         DEAUTHER_DISPLAY // see config.h

         ^

'Button' does not name a type

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

davedarko, 2018-12-06:
have you followed the library steps in the wiki? https://github.com/spacehuhn/esp8266_deauther/wiki/Installation

xwhd12, 2018-12-09:
yes if im not wrong but if u could help me find out whats wrong i would like ur help u could help me tru discord or something else idk but i would like to fix this problem

MartieTheSmartie, 2018-12-13:
You selected Node MCU 0.9, try 1.0 it's a more common device, having issues with compiling aswell..

xwhd12, 2018-12-17:
i did still says it

closed 2019-10-09 by tobozo