SpacehuhnTech/esp8266_deauther #1003: A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

rickwierenga, 2018-12-09:
I'm using esptool.py to flash the firmware and get this error message:

```
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header
```

With Arduino I get

```
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_open failed
```

I know that this is a well known error and I tried many different solutions but none seem to work.

I'm stuck on this chip for a few days without making any progress. 

I'm not very experienced in this field, so please ELI5.

Thanks!

forseenink3238, 2018-12-09:
Press and hold the reset button, while holding the reset button press and
hold the flash button and leave the reset button first then the flash button

tobozo, 2019-10-09:
Closing this issue as OS related, feel free to reopen if necessary.

closed 2019-10-09 by tobozo