SpacehuhnTech/esp8266_deauther #1005: Spi screen bugged

AnnonZerp, 2018-12-11:
I followed the instructions from the wiki to get a screen working and it shows that weird dot pattern with clock cut off in the middle
[image]

marvinborner, 2018-12-11:
This probably happened due to a wrongly configured code. You have to compile it yourself and change the screen parameter in the A_config.h file to your corresponding screen type (In this case you should uncomment the bottom display line for a 1.3" OLED SPI screen I guess).

tobozo, 2018-12-14:
problem with init sequence, typically wrong driver selection

you can try to invert the drivers e.g. if you're using sh1106, try ssd1306 instead

also make sure the pins aren't used by another functionality such as led or button

AnnonZerp, 2018-12-17:
Thanks I mixed up the board type

closed 2018-12-17 by AnnonZerp