SpacehuhnTech/esp8266_deauther #1007: Beacon can not run for a long time?

dydingjiawei, 2018-12-12:
Beacon can not run for a long time?

jLynx, 2018-12-12:
Please give us some more details. We cant do anything off of the information you have given us

spacehuhn, 2018-12-12:
Beacon can not run for a long time?

Beacon can not run for a long time!

jLynx, 2018-12-12:
Well that would be because the standard ESP8266 doesn't have legs. If you add legs to it, it will then hopefully be able to run for a long time. Hope tiis solves your problem

dydingjiawei, 2018-12-13:
When I try to broadcast multiple SSID, it doesn't last long, I need to reload and click on the attack before I can broadcast again.当我尝试广播多个SSID时，它不能持续很久，需要我重新加载再点击攻击才可以再一次广播。

spacehuhn, 2018-12-13:
Maybe this helps:
https://github.com/spacehuhn/esp8266_deauther/wiki/FAQ#attack-stops-after-a-while