SpacehuhnTech/esp8266_deauther #1017: NodeMCU v3 restarting itself

lMartin3, 2018-12-27:
I flashed the deauther using Arduino IDE. It compiled and uploaded successfully but now my nodemcu is restarting itself.
<details>
  <summary> Compile log (...)</summary>
```
⸮⸮⸮H9⸮⸮L⸮⸮⸮L⸮!⸮L⸮)!ť⸮)L⸮SSIDs loaded from /ssids.json

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40253c9d depc=0x00000000

ctx: cont 
sp: 3fff19f0 end: 3fff1c40 offset: 01a0

>>>stack>>>
3fff1b90:  3fff4a88 0000000f 0000000b 3fff4a88  
3fff1ba0:  40253c9d 00000016 00000006 3fff4f80  
3fff1bb0:  00000006 3fff4f80 3fff0770 402093db  
3fff1bc0:  3fff4aa8 0000000f 00000001 3fff2a90  
3fff1bd0:  3fff0910 3fff4cec 00000000 00000000  
3fff1be0:  00000000 3fff0934 3fff0b50 40267488  
3fff1bf0:  3fff1d34 3fff0b50 3fff0934 4021f81f  
3fff1c00:  00000000 00000000 00000000 00000000  
3fff1c10:  00000000 00000000 00000000 3fff0c14  
3fff1c20:  3fffdc20 00000000 3fff0c0d 40226142  
3fff1c30:  00000000 00000000 3fff0c20 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Mounting SPIFFS...OK
Formatting SPIFFS...OK
Switched to Channel 9
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40253c9d depc=0x00000000

ctx: cont 
sp: 3fff19f0 end: 3fff1c40 offset: 01a0

>>>stack>>>
3fff1b90:  3fff4a88 0000000f 0000000b 3fff4a88  
3fff1ba0:  40253c9d 00000016 00000006 3fff4f80  
3fff1bb0:  00000006 3fff4f80 3fff0770 402093db  
3fff1bc0:  3fff4aa8 0000000f 00000001 3fff2a90  
3fff1bd0:  3fff0910 3fff4cec 00000000 00000000  
3fff1be0:  00000000 3fff0934 3fff0b50 40267488  
3fff1bf0:  3fff1d34 3fff0b50 3fff0934 4021f81f  
3fff1c00:  00000000 00000000 00000000 00000000  
3fff1c10:  00000000 00000000 00000000 3fff0c14  
3fff1c20:  3fffdc20 00000000 3fff0c0d 40226142  
3fff1c30:  00000000 00000000 3fff0c20 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Mounting SPIFFS...OK
Formatting SPIFFS...OK
Switched to Channel 9
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40253c9d depc=0x00000000

ctx: cont 
sp: 3fff19f0 end: 3fff1c40 offset: 01a0

>>>stack>>>
3fff1b90:  3fff4a88 0000000f 0000000b 3fff4a88  
3fff1ba0:  40253c9d 00000016 00000006 3fff4f80  
3fff1bb0:  00000006 3fff4f80 3fff0770 402093db  
3fff1bc0:  3fff4aa8 0000000f 00000001 3fff2a90  
3fff1bd0:  3fff0910 3fff4cec 00000000 00000000  
3fff1be0:  00000000 3fff0934 3fff0b50 40267488  
3fff1bf0:  3fff1d34 3fff0b50 3fff0934 4021f81f  
3fff1c00:  00000000 00000000 00000000 00000000  
3fff1c10:  00000000 00000000 00000000 3fff0c14  
3fff1c20:  3fffdc20 00000000 3fff0c0d 40226142  
3fff1c30:  00000000 00000000 3fff0c20 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Mounting SPIFFS...OK
Formatting SPIFFS...
```
</details>
<br><br><br>

lMartin3, 2018-12-27:
Upload settings:
Board: "Generic ESP8266 Module" (From deauther modules)
Upload speed: 115200
CPU Frequency: 160MHz
Flash size: 1M (256K SPIFFS)
Flash mode: DIO
Flash frequency: 80MHz
Upload using: Serial
Reset method: nodemcu

msowa2424, 2018-12-29:
Uploaded using Arduino IDE to Wristband V2. Same problem, no display too.

msowa2424, 2018-12-29:
I tried to upload the code using Arduino IDE on a PC this time and it works. The issue I had is using a Mac.

brycejones, 2018-12-30:
Thanks, I will try windows. I am experiencing the same reset issue using either OSX or Linux.

lMartin3, 2018-12-30:
I tried on Windows and it worked perfectly. The issue I was having was on Linux. I don't know why, I never had problems like these before.

brycejones, 2018-12-31:
Windows worked great. I did not have a reset issue using binaries compiled on Windows.  Thank you for the tip. I have not been successful identifying why we are seeing this reset with binaries compiled on Linux/OSX.

tobozo, 2018-12-31:
@brycejones @lMartin3 would you mind capturing the compile log on linux and specify the Arduino IDE version you were using when the resulting binary fails ?

brycejones, 2019-01-01:
@tobozo - I attached the linux compile log, environment, and serial message including exception and decode. Thank you.

[deauth_linux_reset.log](https://github.com/spacehuhn/esp8266_deauther/files/2719485/deauth_linux_reset.log)

tobozo, 2019-01-01:
@brycejones if it's caused by linux PATH search order, maybe it's worth focusing on this ? 

```
Multiple libraries were found for "Adafruit_NeoPixel.h"
 Not used: /home/brycej/.arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/libraries/Adafruit_NeoPixel-1.1.6
Multiple libraries were found for "ArduinoJson.h"
 Used: /home/brycej/Arduino/libraries/ArduinoJson
 Not used: /home/brycej/.arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/libraries/ArduinoJson
Multiple libraries were found for "SH1106Wire.h"
 Used: /home/brycej/Arduino/libraries/ESP8266_and_ESP32_Oled_Driver_for_SSD1306_display
 Not used: /home/brycej/.arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/libraries/esp8266-oled-ssd1306-4.0.0
 Not used: /home/brycej/.arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/libraries/esp8266-oled-ssd1306-4.0.0
 Not used: /home/brycej/.arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/libraries/esp8266-oled-ssd1306-4.0.0
 Not used: /home/brycej/.arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/libraries/esp8266-oled-ssd1306-4.0.0
```
other possible option: manually upload to SPIFFS before flashing :  

[image]

brycejones, 2019-01-02:
@tobozo -both good ideas. I went thru and modified includes using explicit path to correct libraries and observed reset. I am exploring manually uploading SPIFFS, but believe I need to modify code to prevent SPIFFS from being formatted on first boot. I will persevere. Thank you for your input.

sro2, 2019-01-08:
I flash it by nodmcu flasher. But when i attack 4or 5 device my pown nework connection lost but attack still running background. When i power off my board and plugged again the i can connect pown . Plz give me solution plz.......

closed 2019-10-09 by tobozo