SpacehuhnTech/esp8266_deauther #1023: why its only blinking ???

aponfurious, 2019-01-07:
I changed the attacktimeout to 0. I flash it through arduino IDE software it flash successfully. After flash it only LED light is blinking nothing happen, i also try .bin file by export it from arduino IDE BUT the bin that i download from github is work fine. what is the solution?? 
OR it is possible to send me a deauther .bin file with attacktimeout=0.
if it possible then send it through gmail ..
mail:aponfurious@gmail.com

sro2, 2019-01-07:
Board number send me

<https://github.com/spacehuhn/esp8266_deauther/issues/1023>, or mute the
thread

aponfurious, 2019-01-07:
board no: esp8266
please send me the bin file please..

aponfurious, 2019-01-07:
is no one is here to help me????

closed 2019-01-07 by aponfurious