SpacehuhnTech/esp8266_deauther #1024: Plese send the bin file...

aponfurious, 2019-01-07:
Please send me the bin file of esp8266 board jammer which attacktimeout=0 need it urgent...  please help me... please please please

forseenink3238, 2019-01-07:
Where should we send it?

sro2, 2019-01-07:
I can give but in this bin file i also get a problem. When i attack 3or4
device the deauther wifi connection lost but attack running background. If
you want i can send you bin file. Or you can also give me the Bin file, i
will modify it for you. Thanks

sro2, 2019-01-07:
In mail

sro2, 2019-01-07:
Send dude in gmail

sro2, 2019-01-07:
Which version bin file you are using. And where you live

aponfurious, 2019-01-07:
sorry, I am new user so I don't see the attacktimeout value in setting option..  but now I saw it and solved the problem. thank you everyone

aponfurious, 2019-01-07:
thank you sro2

closed 2019-01-07 by aponfurious