SpacehuhnTech/esp8266_deauther #1026: Why my connection lost when i attack 4or5 device

sro2, 2019-01-08:
> Please search for existing (open and closed) issues first to avoid duplicates.  
Also have a look at the [Wiki](https://github.com/spacehuhn/esp8266_deauther/wiki).  
Please put error messages and code ```inside these 3 quotes/grave accents```
(Delete this text when you've read it)

<details>
  <summary> Compile log (...)</summary>

  
I flash it by nodmcu flasher. But when i attack 4or 5 device my pown nework connection lost but attack still running background. When i power off my board and plugged again the i can connect pown . Plz give me solution plz.......</details>
<br><br><br>

sro2, 2019-01-08:
Please help me

spacehuhn, 2019-01-08:
sir bro plz read manual
https://github.com/spacehuhn/esp8266_deauther/wiki/FAQ#timeout-error
u loose connection when u attack is normal, say it in web interface too where u click

sro2, 2019-01-09:
Sir its happened also web view. Help

sro2, 2019-01-10:
> sir plz read zpazun mezzage plz

Link

closed 2019-10-09 by tobozo