SpacehuhnTech/esp8266_deauther #1030: ERROR: SPIFFS Bootloop

ricolay, 2019-01-11:
hi this is my second time writing this because my computer crashed after i accidentally ran the reset sketch on my laptop and it shut down uhhhhhh crap. I found a NodeMCU ESP8266 thingy in my drawer and wanted to update the deauther on it. I reset it using the Reset_Sketch.ino on Arduino then used the esptoolgui to put the latest (2.1.0) bin on the NodeMCU, it worked, then I downloaded the source files, launched and uploaded the ino file for the deauther, it went through, then I wanted to open the serial monitor to reboot and test it, and it was being spammed with:

<details>
  <summary> Compile log (...)</summary>

```Switched to Channel 9
Settings OK
Switched to Channel 9
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40253c9d depc=0x00000000

ctx: cont 
sp: 3fff1a00 end: 3fff1c50 offset: 01a0

>>>stack>>>
3fff1ba0:  3fff4a98 0000000f 0000000b 3fff4a98  
3fff1bb0:  40253c9d 00000016 00000006 3fff4f90  
3fff1bc0:  00000006 3fff4f90 3fff0780 402093db  
3fff1bd0:  3fff4ab8 0000000f 00000001 3fff2aa0  
3fff1be0:  3fff0920 3fff4cfc 00000000 00000000  
3fff1bf0:  00000000 3fff0944 3fff0b60 40267488  
3fff1c00:  3fff1d44 3fff0b60 3fff0944 4021f81f  
3fff1c10:  00000000 00000000 00000000 00000000  
3fff1c20:  00000000 00000000 00000000 3fff0c24  
3fff1c30:  3fffdc20 00000000 3fff0c1d 40226142  
3fff1c40:  00000000 00000000 3fff0c30 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Mounting SPIFFS...OK
Formatting SPIFFS...OK
Switched to Channel 9
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40253c9d depc=0x00000000

ctx: cont 
sp: 3fff1a00 end: 3fff1c50 offset: 01a0

>>>stack>>>
3fff1ba0:  3fff4a98 0000000f 0000000b 3fff4a98  
3fff1bb0:  40253c9d 00000016 00000006 3fff4f90  
3fff1bc0:  00000006 3fff4f90 3fff0780 402093db  
3fff1bd0:  3fff4ab8 0000000f 00000001 3fff2aa0  
3fff1be0:  3fff0920 3fff4cfc 00000000 00000000  
3fff1bf0:  00000000 3fff0944 3fff0b60 40267488  
3fff1c00:  3fff1d44 3fff0b60 3fff0944 4021f81f  
3fff1c10:  00000000 00000000 00000000 00000000  
3fff1c20:  00000000 00000000 00000000 3fff0c24  
3fff1c30:  3fffdc20 00000000 3fff0c1d 40226142  
3fff1c40:  00000000 00000000 3fff0c30 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Mounting SPIFFS...OK
Formatting SPIFFS...OK
```

</details>
<br><br><br>
I can't do any serial commands and tried used my other NodeMCU thing and doing the same thing and got the same result and now neither work at all. btw after the first loop of the spiff stuff, i can't upload bin files in esptool anymore. I can use the Reset Sketch, but then it just spams the serial with this:

```STARTING...
EEPROM initialized
EEPROM cleaned
SPIFFS initialized
SPIFFS cleaned
WiFi credentials erased
DONE!

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

STARTING...
EEPROM initialized
EEPROM cleaned
SPIFFS initialized
SPIFFS cleaned
WiFi credentials erased
DONE!

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

STARTING...
EEPROM initialized
EEPROM cleaned
SPIFFS initialized
SPIFFS cleaned
WiFi credentials erased
DONE!

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

STARTING...
EEPROM initialized
EEPROM cleaned
SPIFFS initialized
SPIFFS cleaned
WiFi credentials erased
DONE!
```

ricolay, 2019-01-11:
I'll provide any info u need about my model etc.

tobozo, 2019-01-11:
You can actually reflash the ESP if you manage to get out of the boot loop e.g. by keeping the "flash" button pushed while it's rebooting (check your serial console when doing that, expect ).

0x3ff00204b, 2019-01-12:
I'm also having this issue, using NodeMCU. Tried @tobozo 's trick without success. After uploading with flash pushed, nothing happens and i get `wdt reset` at the bottom of serial. No wifi networks found. If I reset the chip, it gets back to the loop.

ricolay, 2019-01-13:
@0x3ff00204b I'll try. Thank you. The flashing trick did not work by the way.

A13k2, 2019-01-31:
I have the same problem/output like @0x3ff00204b with nodemcu v3, 4MB SPIFFS doesn't change anything...

0x3ff00204b, 2019-02-01:
Lower your SPIFFS. I had to set my flash size to 1MB with 256K SPIFFS (probably works with 500K or 1M too) to get it working. 

Also you can try just cloning the repo so you will get the latest features and fixes. That’s what I did.

jumblies, 2019-02-14:
Same problem here with SPIFFS to 256K.  Erase flash did nothing.  I had this problem with a binary too but it resolved.

thomasnilsen, 2019-04-03:
I seem to have the same issue. Tried all possible combinations of flash size / SPIFFS. Just loops over and over again.

`Mounting SPIFFS...OK
Formatting SPIFFS...OK
Switched to Channel 9
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40253c45 depc=0x00000000

ctx: cont 
sp: 3fff1ad0 end: 3fff1d20 offset: 01a0

>>>stack>>>
3fff1c70:  3fff4a98 0000000f 0000000b 3fff4a98  
3fff1c80:  40253c45 00000016 00000006 3fff4f90  
3fff1c90:  00000006 3fff4f90 3fff0780 402093ab  
3fff1ca0:  3fff4ab8 0000000f 00000001 3fff2aa0  
3fff1cb0:  3fff0920 3fff4cfc 00000000 00000000  
3fff1cc0:  00000000 3fff0944 3fff0c30 40267430  
3fff1cd0:  3fff1d44 3fff0c30 3fff0944 4021f7ef  
3fff1ce0:  00000000 00000000 00000000 00000000  
3fff1cf0:  00000000 00000000 00000000 3fff0cf4  
3fff1d00:  3fffdc20 00000000 3fff0ced 402260e2  
3fff1d10:  00000000 00000000 3fff0d00 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Mounting SPIFFS...OK
Formatting SPIFFS...OK
`

lennartkoopmann, 2019-06-08:
FYI I had the same problems on Ubuntu Linux with a Adafruit HUZZAH feather and fixed it by uploading the 1MB .bin file with `esptool` as described in the docs instead of using Arduino IDE.

tobozo, 2019-10-09:
Closing this issue as inactive, feel free to reopen if necessary.

closed 2019-10-09 by tobozo

mihabozic123, 2019-11-01:
Help please!!!
This is my problem:

```
MountingSPIFFS...OK
Formatting SPIFFS...OK
Switched to Channel 9
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled

Exception (29):
epc1=0x4000e1c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: cont
sp: 3fff19d0 end: 3fff1e90 offset: 01a0

>>>stack>>>
3fff1b70:  3fff29d0 00000001 00000001 ffffff80
3fff1b80:  40248ebe 4023c358 00000018 3fff4f78
3fff1b90:  00000001 3fff2968 3fff2968 402490a4
3fff1ba0:  3fff2968 3fff4f78 3fff4c28 00000009
3fff1bb0:  3ffef304 3ffef118 000206cc 3fff4c28
3fff1bc0:  402a3b08 00000064 d63ac66a fffeffff
3fff1bd0:  00000018 0000ffff 00000000 00000001
3fff1be0:  3fff1bf8 4024565c 3ffef094 3fff4f78
3fff1bf0:  d63ac66a 3fffb55b 00000000 00000000
3fff1c00:  00000000 00000000 00000000 00000000
3fff1c10:  00000000 00000000 00000000 00000000
3fff1c20:  00000000 00000000 40246353 3fff4f78
3fff1c30:  3fff0ba4 3fff4f78 3fff4f78 4024639b
3fff1c40:  00000000 3ffef118 3ffe9fbc 00000000
3fff1c50:  40235169 00000002 3fff0ba4 3fff4c28
3fff1c60:  40235281 00000002 00000002 80050014
3fff1c70:  00000001 402352da 00000002 3fff4c28
3fff1c80:  40220694 00000001 00000000 00000010
3fff1c90:  00000010 3fff4c48 3fff0ba4 40220799
3fff1ca0:  3fff4f84 00000001 3fff1ec0 3fff3698
3fff1cb0:  00000008 00000010 00000008 402025ed
3fff1cc0:  3fff35a8 4022612e 3fff4f78 40205e07
3fff1cd0:  3fff1e00 00000001 3fff1ec0 4020302d
3fff1ce0:  3fff35a8 00000001 3fff4f78 402266f5
3fff1cf0:  40226160 3fff4c28 3fff1ec0 40226718
3fff1d00:  3fff1e00 00000001 3fff4f78 40226557
3fff1d10:  3fff1e00 00000001 3fff4c28 40227b24
3fff1d20:  3fff4c58 4010695e 3fff1db0 00000001
3fff1d30:  3fff4bf8 4010695e 3fff1da0 4022ae7a
3fff1d40:  3fff1d90 00000001 3fff1db0 4022aed2
3fff1d50:  3ffe8aa4 0000000a 3fff1db0 4022af02
3fff1d60:  40235102 3fff1e10 3fff1e1c 3fff0ba4
3fff1d70:  40234f6d 00000001 3fff1da0 4022a579
3fff1d80:  00000000 3fff1e10 3fff1e1c 402208be
3fff1d90:  0104a8c0 00ffffff 0104a8c0 3ffefdfc
3fff1da0:  3fff0ba4 3fff1e1c 3fff1e10 4021d35c
3fff1db0:  3ffe9758 00ffffff 00000001 4022ae7a
3fff1dc0:  3ffe9758 0104a8c0 3fff1e10 3ffe9758
3fff1dd0:  0104a8c0 7fffffff 3fff1e10 4022af02
3fff1de0:  00000000 00000001 00000000 4022af50
3fff1df0:  3fff3088 3fff0da4 3fff1e30 4026796c
3fff1e00:  3fff08f0 3fff0da4 3fff0ab4 4021d9e5
3fff1e10:  3fff4c48 0000000f 00000008 3fff4c28
3fff1e20:  0000000f 00000005 3fff4c08 0000000f
3fff1e30:  00000004 0000000f 00000000 40209290
3fff1e40:  3fff1eb4 3fff0da4 3fff0ab4 4021f9b1
3fff1e50:  00000000 00000000 00000000 00000000
3fff1e60:  00000000 00000000 00000000 3fff0e68
3fff1e70:  3fffdc20 00000000 3fff0e61 40226296
3fff1e80:  00000000 00000000 3fff0e70 40100114
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16
tail 0
chksum 0x42
csum 0x42
~ld
Mounting SPIFFS...OK
Formatting SPIFFS...OK
Switched to Channel 9
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled

Exception (29):
epc1=0x4000e1c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: cont
sp: 3fff19d0 end: 3fff1e90 offset: 01a0

>>>stack>>>
3fff1b70:  3fff29d0 00000001 00000001 ffffff80
3fff1b80:  40248ebe 4023c358 00000018 3fff4f78
3fff1b90:  00000001 3fff2968 3fff2968 402490a4
3fff1ba0:  3fff2968 3fff4f78 3fff4c28 00000009
3fff1bb0:  3ffef304 3ffef118 000206cc 3fff4c28
3fff1bc0:  402a3b08 00000064 d63ac66a fffeffff
3fff1bd0:  00000018 0000ffff 00000000 00000001
3fff1be0:  3fff1bf8 4024565c 3ffef094 3fff4f78
3fff1bf0:  d63ac66a 3fffb55b 00000000 00000000
3fff1c00:  00000000 00000000 00000000 00000000
3fff1c10:  00000000 00000000 00000000 00000000
3fff1c20:  00000000 00000000 40246353 3fff4f78
3fff1c30:  3fff0ba4 3fff4f78 3fff4f78 4024639b
3fff1c40:  00000000 3ffef118 3ffe9fbc 00000000
3fff1c50:  40235169 00000002 3fff0ba4 3fff4c28
3fff1c60:  40235281 00000002 00000002 80050014
3fff1c70:  00000001 402352da 00000002 3fff4c28
3fff1c80:  40220694 00000001 00000000 00000010
3fff1c90:  00000010 3fff4c48 3fff0ba4 40220799
3fff1ca0:  3fff4f84 00000001 3fff1ec0 3fff3698
3fff1cb0:  00000008 00000010 00000008 402025ed
3fff1cc0:  3fff35a8 4022612e 3fff4f78 40205e07
3fff1cd0:  3fff1e00 00000001 3fff1ec0 4020302d
3fff1ce0:  3fff35a8 00000001 3fff4f78 402266f5
3fff1cf0:  40226160 3fff4c28 3fff1ec0 40226718
3fff1d00:  3fff1e00 00000001 3fff4f78 40226557
3fff1d10:  3fff1e00 00000001 3fff4c28 40227b24
3fff1d20:  3fff4c58 4010695e 3fff1db0 00000001
3fff1d30:  3fff4bf8 4010695e 3fff1da0 4022ae7a
3fff1d40:  3fff1d90 00000001 3fff1db0 4022aed2
3fff1d50:  3ffe8aa4 0000000a 3fff1db0 4022af02
3fff1d60:  40235102 3fff1e10 3fff1e1c 3fff0ba4
3fff1d70:  40234f6d 00000001 3fff1da0 4022a579
3fff1d80:  00000000 3fff1e10 3fff1e1c 402208be
3fff1d90:  0104a8c0 00ffffff 0104a8c0 3ffefdfc
3fff1da0:  3fff0ba4 3fff1e1c 3fff1e10 4021d35c
3fff1db0:  3ffe9758 00ffffff 00000001 4022ae7a
3fff1dc0:  3ffe9758 0104a8c0 3fff1e10 3ffe9758
3fff1dd0:  0104a8c0 7fffffff 3fff1e10 4022af02
3fff1de0:  00000000 00000001 00000000 4022af50
3fff1df0:  3fff3088 3fff0da4 3fff1e30 4026796c
3fff1e00:  3fff08f0 3fff0da4 3fff0ab4 4021d9e5
3fff1e10:  3fff4c48 0000000f 00000008 3fff4c28
3fff1e20:  0000000f 00000005 3fff4c08 0000000f
3fff1e30:  00000004 0000000f 00000000 40209290
3fff1e40:  3fff1eb4 3fff0da4 3fff0ab4 4021f9b1
3fff1e50:  00000000 00000000 00000000 00000000
3fff1e60:  00000000 00000000 00000000 3fff0e68
3fff1e70:  3fffdc20 00000000 3fff0e61 40226296
3fff1e80:  00000000 00000000 3fff0e70 40100114
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16
tail 0
chksum 0x42
csum 0x42
~ld

```
(and the loop continues)

TheMiguelBi, 2019-12-17:
I also have this problem (I use WemosD1mini). 
So using ArduinoIDE doesn't work? 
are you saying the problem is this?

I have tried several combinations of Spiffs memory... :/