SpacehuhnTech/esp8266_deauther #1033: ERROR: you guessed it the ArduinoJson error

FallenCancer, 2019-01-15:
So basically I do have the correct version but it is not working what so ever
https://gyazo.com/23b0384f528e72170716aa9635f52ada <-- proof
and this keeps popping up even though I reinstalled Arduino and everything
Nodemcu Lua Amica Module V2 ESP8266 ESP 12E WiFi WiFi Development Board with CP2102
thats the board I have

WARNING: Category '' in library ArduinoJson is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library ArduinoJson is not valid. Setting to 'Uncategorized'
esp8266_deauther:25:2: error: #error Please upgrade/downgrade ArduinoJSON library to version 5!

 #error Please upgrade/downgrade ArduinoJSON library to version 5!

  ^

Multiple libraries were found for "ArduinoJson.h"
 Used: C:\Users\cante\OneDrive\Documents\Arduino\libraries\libraries\ArduinoJson
 Not used: C:\Users\cante\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\ArduinoJson
exit status 1
#error Please upgrade/downgrade ArduinoJSON library to version 5!

per1234, 2019-01-15:
This warning:
```
WARNING: Category '' in library ArduinoJson is not valid. Setting to 'Uncategorized'
```
Indicates you're using a very outdated version of the ArduinoJson library (5.0.5 or older). Try installing version 5.13.4.

FallenCancer, 2019-01-15:
Alright testin it now thanks g

FallenCancer, 2019-01-15:
CLOSE THIS BITCH

closed 2019-01-15 by FallenCancer