SpacehuhnTech/esp8266_deauther #1040: Default attack on boot

RedBlaze42, 2019-01-20:
Hello,

Do you know how to automaticaly start an attack on boot without having to use a phone or an oled screen ?
I'd like to start a beacon attack or a deauth attack on all channels when I plug the power. Is that possible ?
Anyway thank you for your awesome work.

DeMaska, 2019-01-21:
You have to write commands to the file autostart.txt

If you are using the Arduino Monitor, the command should lool like this :

write autostart.txt scan -ap -c 1min
write autostart.txt enable random 10s
write autostart.txt attack -b -da -t 0

This file automatically executes on each boot. The script above will scan for APs every 1 minute, generate random beacon names every 10 seconds and attack all stations and broadcast beacons indefinitely.

vuong244, 2019-01-24:
You can refer to this post https://www.hackster.io/234v244/deauthentication-attack-with-esp8266-4ed3d3

liklev, 2019-05-17:
hi laurentbourassa,
what commands/syntax i should put into that autostart.txt to scan both ap/sta and do deauth on all discovered item.
then repeat these till i shot down the modul. (similar to kali/mdk3).
is there any detailed wiki of possible on boot commands?
regards

liklev, 2019-05-18:
hi again laurentbourassa,
unfortunately i made noise before proper reading of the wiki.
i found all the good explanations there.
all the best, lev

tobozo, 2019-10-09:
Closing this issue as inactive, feel free to reopen if necessary.

closed 2019-10-09 by tobozo

Hirameki11, 2021-11-16:
> You have to write commands to the file autostart.txt.
> 
> If you are using the Arduino Monitor, the command should look like this :
> 
> write autostart.txt scan -ap -c 1min write autostart.txt enable random 10s write autostart.txt attack -b -da -t 0
> 
> This file automatically executes on each boot. The script above will scan for APs every 1 minute, generate random beacon names every 10 seconds and attack all APs and broadcast beacons indefinitely.
> 
> If you want to see what is contained in that file, simply enter the following command :
> 
> print autostart.txt
> 
> Hope that helps!

Good day sir, I am no programmer myself Im just familiar with coding/scripting. Please help me how to make script to Autostart Deauth Attack to a specific MAC or SSID and momentarily stop the Deauth Attack every 5minutes. So that every time I power on the board I dont need to go to web ui to manually start it. Is it possible to do this sir? Thanks!