SpacehuhnTech/esp8266_deauther #1042: A_config.h:7:24: error: SH1106Wire.h: No such file or directory

yousifja, 2019-01-21:
Arduino: 1.8.7 (Windows 7), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

In file included from sketch\Settings.h:11:0,

                 from C:\Users\ttt\Desktop\esp8266_deauther-master\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino:27:

A_config.h:7:24: error: SH1106Wire.h: No such file or directory

 #include "SH1106Wire.h"

                        ^

compilation terminated.

exit status 1
SH1106Wire.h: No such file or directory

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

itsjustme, 2019-01-31:
make sure you selected you're Board under "ESP8266 Deauther Modules"

sarptekbas, 2019-05-06:
> make sure you selected you're Board under "ESP8266 Deauther Modules"

thanks

tobozo, 2019-10-09:
Closing this issue as inactive, feel free to reopen if necessary.

closed 2019-10-09 by tobozo

sabidkaral, 2019-11-01:
I have the some problem... i already tried out all kinds of esp8266 boards but still have the error

edeavila, 2021-12-10:
> I have the some problem... i already tried out all kinds of esp8266 boards but still have the error

same

Fearce1, 2023-07-12:
The FILE IS MISSING. I've had this problem ALONG WITH MANY others after attempting to "UPDATE" a Dstike OLED MINI V2. The web-interface used to work FINE then I tried to upgrade from the original version it was shipped with (thinking UPGRADE MEANS BETTER) and get a rabbithole of MISSING FILES and LOST FUNCTIONALITY. Seems very intentional for all of these things to suddenly not be in the DOWNLOADABLE SOURCE(S) of each of these versions. The file is NOT INCLUDED in the SOURCE .zip You can DL version 2 that calls the file from the settings.h file. Below shows the .h files and "SH1106Wire.h" IS NOT IN THERE.
[image]
[image]

Seems like it was LEFT OUT of the downloadable files... WHY?

Fearce1, 2023-07-12:
The reason I even attempted a update is so the WEB interface would load the SSID's and NAMES. Every WORKING version throws this error:
ERROR: saving file. Try 'format' and restart - /scan.json
# save names
ERROR: saving file. Try 'format' and restart - /names.json
# format -names.json
Formatting SPIFFS...OK
# save scan
ERROR: saving file. Try 'format' and restart - /scan.json
# save names
ERROR: saving file. Try 'format' and restart - /names.json