SpacehuhnTech/esp8266_deauther #1052: Can't remove an AP using serial command

29G, 2019-01-29:
Using the serial monitor:
I type "scan -ap" and it shows me a list of the WiFis around me
Let's say i see a network that i do not want to deauth, called pwned, and i try to remove it with the "remove" command.
I try "remove -ap pwned" and i get "Removed access point [a different AP than pwned]"
I try "remove -n pwned" and i get "No device found with name 0" (?! i typed pwned, not 0.)
I try "remove -ss pwned" and i get "Removed SSID Never gonna tell a lie" (wtf?)
Please help me, thanks

DeMaska, 2019-01-31:
Instead of removing it by name, remove it by it's associated numer. Ex :

remove -ap 0

In my experience (I could be wrong), "remove -n [name]" will only work if the AP or station is saved in your known devices and associated to a name of your choice.

So if you add "pwned" to your saved devices, you can put a custom name to it and use the command "remove -n [customName]".

DeMaska, 2019-02-02:
Sorry, I meant to say "deselect -n [customName]". That command should work.

29G, 2019-02-02:
[Screenshot](https://i.imgur.com/DBFDuIp.png)
I tried that but it doeosn't do anything. When i do "show -ap" the network is still there

DeMaska, 2019-02-02:
You misunderstand how it works. You won't be able to not see a scanned network, but you can unselect it (As it is in your picture).

Try this :
scan -ap (It will update all scanned APs)
show -ap (You will see all scanned APs)
select -a (It will select all scanned APs)
show -ap (This time, you should see an * in the "selected" section)
unselect -n LOL (This should unselect the AP named LOL)
show -ap (Now you should not see an * for LOL, but you should see one for all other APs)

I also invite you to look up in the [wiki](https://github.com/spacehuhn/esp8266_deauther/wiki).

29G, 2019-02-02:
Finally it worked, thank you very much for helping

closed 2019-02-02 by 29G