SpacehuhnTech/esp8266_deauther #1059: How to return to default fimware nodemcu?

RUBIKOF, 2019-02-06:
Hello, install fimware ("esp8266_deauther_1mb.bin") in my nodemcu 1.0 (ESP-12E Module) to test it and it works perfectly, but now I would like to return to the normality of my module to load other projects, how do I do this? Could you help me? :(

When trying to load my sketch from the arduino ide process is very slow and when it reaches 65% I get the following error "An error occurred while sending the sketch"
I was inventing and come to this link: https://github.com/spacehuhn/esp8266_deauther/tree/master/Reset_Sketch

I thought that resetting the module was solved but I could not solve it, I still can not load my sketchs

It is worth mentioning that if I generate the binary files of my sketch if I can flash it with the NODEMCU FIMWARE PROGRAMMER (this program is the one I use to flash the file esp8266_deauther_1mb.bin)

tobozo, 2019-10-09:
Closing this issue as inactive, feel free to reopen if necessary.

closed 2019-10-09 by tobozo

spacehuhn, 2020-06-16:
For future reference: there's no default firmware for the ESPs. Maybe there's a NodeMCU firmware for lua you could flash it with.
If you get upload errors it could be a cable, connector or driver issue. But usually you just flash something new on it, even if that's just an empty binary full of zeros to erase the memory without needing new firmware.