SpacehuhnTech/esp8266_deauther #1065: DSTIKE flash

t0tch, 2019-03-01:
After I bought my DSTIKE oled mini off the official DSTIKE aliexpress and receiving the board in full working condition, I wanted to change the attack time out to 0( no timeout). I flash following the tutorial and nothing. I reset following reset sketch nothing. I follow different methods flashing different bins and no screen turn on no led turn on except the connection to pc, its like its bricked

t0tch, 2019-03-01:
If anyone has advice or even a compiled bin for DSTIKE mini v1 with attacktimeout 0 I would be very greatful

tobozo, 2019-03-01:
If you're not familiar with flashing esp8266 you may find the easier path by testing basic examples from beginners tutorials first (e.g. led blink), then gradually add complexity to your examples until you understand every details mentioned in the [installation wiki](https://github.com/spacehuhn/esp8266_deauther/wiki/Installation).

happy hacking ^_^

t0tch, 2019-03-02:
Thanks for the info, I looked at tutorials initially but as soon as I flashed the board the first time I could not get a response anymore and think I flashed it wrong. I'm not currently looking for a way to program the board at the minute just Flash the deauth bin onto the board with some setting s changed with the screen working like before but I have tried esptool and the GUI version and arduino ide and still nothing.
Thanks

labomb, 2019-03-04:
Try changing the flash mode to DOUT instead of the default DIO mode.

I needed to do that with the NodeMCU-07 board. Not sure why it isn't documented somewhere.

t0tch, 2019-03-05:
Thanks @labomb this worked perfectly for my esp-07 and I think it should be documented but not sure if it is but thanks again

closed 2019-03-05 by t0tch