SpacehuhnTech/esp8266_deauther #1068: Used ESP-WROOM -02 disconnected my iPhone, but all my Wemos Sensors ware still connected

ovisopa, 2019-03-06:
Hello everyone.

I just tried now the DEAUTHER on my own network (I selected my Office AP - a TP-link AC1200
Archer C6 Router), as I have more than 10 wemos sensors around my house (smoke sensors, light switches, temperature & humidity sensors, etc) I'm quite concerned about this wifi vulnerability, if it's so easy to mess a network than maybe I should look for another way of connecting my sensors (is LoRa safer?). My security cameras are all wired, with this DEAUTHER it's also possible to disconnect the wireless security cameras from the router?

My iPhone disconnected immediately after I pressed the start button, but even after a few minutes, my office light still worked (I tried a few ON / OFF cycles), also my office smoke monitor was still online. 
I'm curious why my  ESP8266 sensors/switches ware still connected ?

Any ideea ? 

Thank you.

ovisopa, 2019-03-07:
Nobody around ? I'm really curious why all my ESP8266 sensors ware not affected by the DEAUTH ATTACK.

All ESP8266 and iPhones and laptops ware connected to the same SSID, the attack worked perfect on iPhone and Laptops but not on ESP8266 

Thank you.

sro2, 2019-03-07:
Tell the version code

ovisopa, 2019-03-07:
I downloaded the BIN directly: deauther_2.1.0_DSTIKE_OLED_v1_v1.5.bin

sro2, 2019-03-07:
Which device you are using

ovisopa, 2019-03-07:
ESP-WROOM -02 with OLED

https://www.banggood.com/Wemos-D1-Esp-Wroom-02-Motherboard-ESP8266-Mini-WiFi-NodeMCU-Module-p-1224577.html?rmmds=search&cur_warehouse=CN

sro2, 2019-03-07:
I think you have to flash it again ... I used esp8266 without o led ..... I
can attack on any nodemcu and any wairless sensors...... My version is
2.1.0 1mb bin . Use nodmcu flasher .. try it again

closed 2019-10-09 by tobozo