SpacehuhnTech/esp8266_deauther #1069: Crashes on first use

Bougakov, 2019-03-06:
Hi. I have tried to upload the code to Ai-Tinker ESP8266, used three different chips, but get the same problem. It tries to format the SPIFFS and crashes

<details>
  <summary> Compile log (...)</summary>

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/sanja/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/sanja/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/sanja/Documents/Arduino/libraries -fqbn=deauther:esp8266:nodemcuv2:CpuFrequency=160,UploadTool=esptool,FlashFreq=80,FlashMode=dio,UploadSpeed=115200,FlashSize=4M3M -vid-pid=1A86_7523 -ide-version=10900 -build-path /var/folders/cj/y9dvvn9553q4cnxtkhxd16kr0000gn/T/arduino_build_925786 -warnings=none -build-cache /var/folders/cj/y9dvvn9553q4cnxtkhxd16kr0000gn/T/arduino_cache_162926 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=/Users/sanja/Library/Arduino15/packages/esp8266/tools/esptool/0.4.6 -prefs=runtime.tools.esptool-0.4.6.path=/Users/sanja/Library/Arduino15/packages/esp8266/tools/esptool/0.4.6 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/sanja/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -prefs=runtime.tools.xtensa-lx106-elf-gcc-1.20.0-26-gb404fb9-2.path=/Users/sanja/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=/Users/sanja/Library/Arduino15/packages/esp8266/tools/mkspiffs/0.1.2 -prefs=runtime.tools.mkspiffs-0.1.2.path=/Users/sanja/Library/Arduino15/packages/esp8266/tools/mkspiffs/0.1.2 -verbose -experimental /Users/sanja/Downloads/esp8266_deauther-2.1.0/esp8266_deauther/esp8266_deauther.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/sanja/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/sanja/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/sanja/Documents/Arduino/libraries -fqbn=deauther:esp8266:nodemcuv2:CpuFrequency=160,UploadTool=esptool,FlashFreq=80,FlashMode=dio,UploadSpeed=115200,FlashSize=4M3M -vid-pid=1A86_7523 -ide-version=10900 -build-path /var/folders/cj/y9dvvn9553q4cnxtkhxd16kr0000gn/T/arduino_build_925786 -warnings=none -build-cache /var/folders/cj/y9dvvn9553q4cnxtkhxd16kr0000gn/T/arduino_cache_162926 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=/Users/sanja/Library/Arduino15/packages/esp8266/tools/esptool/0.4.6 -prefs=runtime.tools.esptool-0.4.6.path=/Users/sanja/Library/Arduino15/packages/esp8266/tools/esptool/0.4.6 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/sanja/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -prefs=runtime.tools.xtensa-lx106-elf-gcc-1.20.0-26-gb404fb9-2.path=/Users/sanja/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=/Users/sanja/Library/Arduino15/packages/esp8266/tools/mkspiffs/0.1.2 -prefs=runtime.tools.mkspiffs-0.1.2.path=/Users/sanja/Library/Arduino15/packages/esp8266/tools/mkspiffs/0.1.2 -verbose -experimental /Users/sanja/Downloads/esp8266_deauther-2.1.0/esp8266_deauther/esp8266_deauther.ino
Using board 'nodemcuv2' from platform in folder: /Users/sanja/Library/Arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther
Using core 'esp8266' from platform in folder: /Users/sanja/Library/Arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther
Detecting libraries used...
/Users/sanja/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Users/sanja/Library/Arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/tools/sdk/include -c -Os -g -fno-threadsafe-statics -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10900 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266 -DESP8266 -I/Users/sanja/Library/Arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/cores/esp8266 -I/Users/sanja/Library/Arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/variants/nodemcu /var/folders/cj/y9dvvn9553q4cnxtkhxd16kr0000gn/T/arduino_build_925786/sketch/esp8266_deauther.ino.cpp -o /dev/null
/Users/sanja/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Users/sanja/Library/Arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/tools/sdk/include -c -Os -g -fno-threadsafe-statics -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10900 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266 -DESP8266 -I/Users/sanja/Library/Arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/cores/esp8266 -I/Users/sanja/Library/Arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/variants/nodemcu -I/Users/sanja/Library/Arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/libraries/EEPROM /var/folders/cj/y9dvvn9553q4cnxtkhxd16kr0000gn/T/arduino_build_925786/sketch/esp8266_deauther.ino.cpp -o /dev/null
/Users/sanja/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Users/sanja/Library/Arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/tools/sdk/include -c -Os -g -fno-threadsafe-statics -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10900 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266 -DESP8266 -I/Users/sanja/Library/Arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/cores/esp8266 -I/Users/sanja/Library/Arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/variants/nodemcu -I/Users/sanja/Library/Arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/libraries/EEPROM -I/Users/sanja/Documents/Arduino/libraries/ArduinoJson/src /var/folders/cj/y9dvvn9553q4cnxtkhxd16kr0000gn/T/arduino_build_925786/sketch/esp8266_deauther.ino.cpp -o /dev/null
/Users/sanja/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Users/sanja/Library/Arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/tools/sdk/include -c -Os -g -fno-threadsafe-statics -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10900 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266 -DESP8266 -I/Users/sanja/Library/Arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/cores/esp8266 -I/Users/sanja/Library/Arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/variants/nodemcu -I/Users/sanja/Library/Arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/libraries/EEPROM -I/Users/sanja/Documents/Arduino/libraries/ArduinoJson/src -I/Users/sanja/Library/Arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/libraries/Wire /var/folders/cj/y9dvvn9553q4cnxtkhxd16kr0000gn/T/arduino_build_925786/sketch/esp8266_deauther.ino.cpp -o /dev/null
/Users/sanja/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Users/sanja/Library/Arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/tools/sdk/include -c -Os -g -fno-threadsafe-statics -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10900 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266 -DESP8266 -I/Users/sanja/Library/Arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/cores/esp8266 -I/Users/sanja/Library/Arduino15/packages/ [... cut ...]
3fff1c10:  00000000 00000000 00000000 00000000  
3fff1c20:  00000000 00000000 00000000 3fff0c24  
3fff1c30:  3fffdc20 00000000 3fff0c1d 402264be  
3fff1c40:  00000000 00000000 3fff0c30 40100114  
stack

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Mounting SPIFFS...OK
```

Power is definitely not an issue. The board is AI-Tinker's 8266MOD, I am using "Nodemcu 1.1" option from provided list of boards as "Generic ESP8266" gives me an error `warning: espcomm_sync failed`

tobozo, 2019-03-06:
`FlashSize=4M3M` maybe your ESP's flash isn't that big, have you tried with other values (e.g. 1M/512K SPIFFS) ?

Bougakov, 2019-03-06:
It is definitely 4Mb, yet I tried other options, starting from 1Mb.

tobozo, 2019-03-06:
did you try pre-formatting your spiffs ?

mgx0, 2019-03-12:
did you compile that on Macintosh Arduino?

Bougakov, 2019-03-13:
Thanks for the hint, I'll give Windows Arduino a try.

xhdix, 2019-04-05:
@Bougakov 
Please decode the exception with this tool:
https://github.com/me-no-dev/EspExceptionDecoder

tobozo, 2019-10-09:
Closing this issue as inactive, feel free to reopen if necessary.

closed 2019-10-09 by tobozo

Zuclo, 2019-11-25:
I have the same issue and tracked it to the "cli.load();" call. then it crashes everytime.
I removed cli.load(); and now works as intended. 
I'm not sure what cli actually does, but it works without it.

potassium-chloride, 2020-07-03:
I check cli.load() and didn't found file "autostart.txt" in deauther project. I deleted cli.load() body and ESP starts normal.