SpacehuhnTech/esp8266_deauther #1077: Pwned network not showing

thebossjd, 2019-04-01:
So I tried with different versions and various flashers and even compiled with arduino and it uploads fine. If it's plugged the light blinks so it works. However, the wifi doesn't show (looked both on my phone and pc). I see there's many people with this problem on here but no conclusive answers. 

My model is a Nodemcu ESP8266. OS Windows 10.

Any ideas? Thanks in advance.

Dizeeee, 2019-04-05:
If you are using Arduino try changing the flash size to 4M

luke666, 2019-04-15:
I've managed to get v2.1.0 somehow running (loses connection sometimes or throws a timeout when browsing the web UI) on this board: https://robotdyn.com/wifi-nodem-esp8266-32m-flash-ch340g.html - This board supports v 1.6 without problems, 2.x versions were not displaying essid or not working at all. If somebody have the same board, try to erase the flash completely first:
esptool.py erase_flash
and then write the image with:
esptool.py write_flash 0x00000 deauther_2.1.0_4mb.bin
then play for some time with the board over minicom (115200 8N1), try to change channel, password, turn off autosave.

TheProgrammingM, 2019-05-19:
Same problem here. I tried those comments, but it  doesen't work. I can't see the network 'pwned'. The module is working, i uploaded a test sketch.

closed 2019-10-09 by tobozo