SpacehuhnTech/esp8266_deauther #1080: Precompiled BIN for SPI-Display needed - otherwise boot loop

Einstein2150, 2019-04-08:
I tried to compile with Arduino IDE but it causes a boot-loop. **The recompiled BIN worked fine but it doesn't support SPI displays. Can you precompile a configuration with SPI display enabled?** All precompiled one seems to use i2c connections :-(

PS: I tried also naming the pins by "D1", "D2" and so on - nothing changed :-/

By the way - here is the boot loop:

```
Mounting SPIFFS...OK
Formatting SPIFFS...OK
Switched to Channel 9
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40253aad depc=0x00000000

ctx: cont 
sp: 3fff1ad0 end: 3fff1d20 offset: 01a0

>>>stack>>>
3fff1c70:  3fff5968 0000000f 0000000b 3fff5968  
3fff1c80:  40253aad 00000016 00000006 3fff60d8  
3fff1c90:  00000006 3fff60d8 3fff0780 4020901b  
3fff1ca0:  3fff5988 0000000f 00000001 3fff06e4  
3fff1cb0:  3fff0920 3fff5e44 00000000 00000000  
3fff1cc0:  00000000 3fff0944 3fff0c34 40267298  
3fff1cd0:  3fff05a4 3fff0c34 3fff0944 4021f453  
3fff1ce0:  00000000 00000000 00000000 00000000  
3fff1cf0:  00000000 00000000 00000000 3fff0cf8  
3fff1d00:  3fffdc20 00000000 3fff0cf1 40225f4a  
3fff1d10:  00000000 00000000 3fff0d00 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

```

tobozo, 2019-04-08:
you'll probably have to edit the code yourself to get this working, what display model is that ?

Einstein2150, 2019-04-08:
That is the Part: € 1,62  6％ Rabatt | 0,96 zoll OLED Weiß Display Modul 128X64 OLED SPI 7pin Fahrer Chip SSD1306 für arduino Diy Kit
https://s.click.aliexpress.com/e/XewzISM

Einstein2150, 2019-04-09:
OK. The display is working using this Display-Test-Code:

```
/**************************************************************************
 This is an example for our Monochrome OLEDs based on SSD1306 drivers

 Pick one up today in the adafruit shop!
 ------> http://www.adafruit.com/category/63_98

 This example is for a 128x64 pixel display using SPI to communicate
 4 or 5 pins are required to interface.

 Adafruit invests time and resources providing this open
 source code, please support Adafruit and open-source
 hardware by purchasing products from Adafruit!

 Written by Limor Fried/Ladyada for Adafruit Industries,
 with contributions from the open source community.
 BSD license, check license.txt for more information
 All text above, and the splash screen below must be
 included in any redistribution.
 **************************************************************************/

#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels

// Declaration for SSD1306 display connected using software SPI (default case):
#define OLED_MOSI   13
#define OLED_CLK   14
#define OLED_DC    4
#define OLED_CS    15
#define OLED_RESET 5
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT,
  OLED_MOSI, OLED_CLK, OLED_DC, OLED_RESET, OLED_CS);

/* Comment out above, uncomment this block to use hardware SPI
#define OLED_DC     6
#define OLED_CS     7
#define OLED_RESET  8
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT,
  &SPI, OLED_DC, OLED_RESET, OLED_CS);
*/

#define NUMFLAKES     10 // Number of snowflakes in the animation example

#define LOGO_HEIGHT   16
#define LOGO_WIDTH    16
static const unsigned char PROGMEM logo_bmp[] =
{ B00000000, B11000000,
  B00000001, B11000000,
  B00000001, B11000000,
  B00000011, B11100000,
  B11110011, B11100000,
  B11111110, B11111000,
  B01111110, B11111111,
  B00110011, B10011111,
  B00011111, B11111100,
  B00001101, B01110000,
  B00011011, B10100000,
  B00111111, B11100000,
  B00111111, B11110000,
  B01111100, B11110000,
  B01110000, B01110000,
  B00000000, B00110000 };

[... 420 lines cut ...]
//#define BUTTON_LEFT 12
//#define BUTTON_RIGHT 13
//#define BUTTON_B 10

/* = = = = = = = = = = = = = = = = = = = = = = = = = = = = = =
    Uncomment the type of display you're using
    Only one of them can be defined at a time!
   = = = = = = = = = = = = = = = = = = = = = = = = = = = = = =
*/
//#define DEAUTHER_DISPLAY SSD1306Wire display = SSD1306Wire(0x3c, 5, 4); // for 0.96" OLED
//#define DEAUTHER_DISPLAY SH1106Wire display = SH1106Wire(0x3c, 5, 4); // for 1.3" OLED

/*  RST = GPIO 5 (D1)
    DC = GPIO 4 (D2)
    CS = GPIO 15 (D8) or GND
    SCK/SCL = GPIO 14 (D5)
    SDA/MOSI = GPIO 13 (D7) */
#define DEAUTHER_DISPLAY SSD1306Spi display = SSD1306Spi(5, 4, 15); // for 0.96" OLED with SPI
//#define DEAUTHER_DISPLAY SH1106Spi display = SH1106Spi(5, 4, 15); // for 1.3" OLED with SPI

//#define FLIP_DIPLAY // uncomment that to flip the display vertically
// ========================================================= //

#endif
```

tobozo, 2019-04-09:
now you have the correct pinout setup, you could compare with the assignations found in the config files of this project and see if there are conflicts, you may have to reassign new pins for `Button A` and `Button Down` as those seem to use 13 and 14 too.

```
#define OLED_MOSI   13
#define OLED_CLK   14
#define OLED_DC    4
#define OLED_CS    15
#define OLED_RESET 5
```

Einstein2150, 2019-04-09:
Great. The Button-Pin-Layout was the Problem. Display works :-)

closed 2019-04-09 by Einstein2150