SpacehuhnTech/esp8266_deauther #1084: Adding MPR121 i2c touch module instead of buttons (helpRequest)

gibranZawahra, 2019-04-14:
i'm trying to add an mpr121 touch module instead of the 6 buttons recommended, is this possible hardware and code wise? 
and is the "DisplayUI.cpp" the only file i need to modify ?
please advise and sorry if this is not where i'm supposed to ask, first issue ever!
thanks a bunch!

tobozo, 2019-04-14:
MPR21 seems to be an i2c device so it should be sharing the i2c bus with the OLED display without any issue. 

Suggested turing path:

 - get OLED and Touch working without the deauther (create a new sketch, re-use the OLED driver)
 - add touch driver to the deauther
 - edit deauther logic to listen to touch instead of buttons

gibranZawahra, 2019-04-14:
thanks ! i'm already almost there, the module is initializing correctly but i'm lost in the files looking for the function to modify, i'm looking for the one that sort digitalRead() the buttons to give the values of the touch buttons, could kindly point to the correct function ? 

thanks a lot!

tobozo, 2019-04-14:
It's using the [SimpleButton](https://github.com/spacehuhn/SimpleButton) library, you'll probably need to add the "Touch" and "TouchRelease" events

spacehuhn, 2019-04-14:
Let me know how it goes.
I will order a MP121 to test it out. I'm currently working on cleaning up the SimpleButton library.

gibranZawahra, 2019-04-14:
it seems i'm going backwards, the module isn't initializing although the i2c scanner shows its connected correctly, i think it's because the begin function of the mpr121 is not using the coorrect pins for i2c, for the ssd1306 display you would specify the address and the pins, but in the mpr121.begin you only 
specify the address, i'm looking for a way to give the precise pins.
this is the first part where i try to start the i2c:
`#include "DisplayUI.h"
#include <Wire.h>
#include "Adafruit_MPR121.h"

Adafruit_MPR121 cap = Adafruit_MPR121();

// ===== adjustable ===== //
void DisplayUI::configInit() {
  
  // initialize display
  Serial.println("init Display");
  if(cap.begin(0x5a)){
    Serial.println("MPR121 Started");
  }
  display.init();
`
and this is a test function that only prints the currently touched electrode
`void DisplayUI::updateTouch()
{
  uint16_t lasttouched = 0;
  uint16_t currtouched = 0;
  currtouched = cap.touched();
 
  for (uint8_t i = 0; i < 12; i++) {
    // it if *is* touched and *wasnt* touched before, alert!
    if ((currtouched & _BV(i)) && !(lasttouched & _BV(i)) ) {
      Serial.print(i); Serial.println(" touched");
    }
    // if it *was* touched and now *isnt*, alert!
    if (!(currtouched & _BV(i)) && (lasttouched & _BV(i)) ) {
      Serial.print(i); Serial.println(" released");
    }
  }`
i call it in the deauther loop..
thanks tobozo for your help and spacehuhn for all the effort you put into this project!

gibranZawahra, 2019-04-14:
ok so this is a weird way but i managed to start the i2c comms using the i think deprecated wire.pins(5,4);
function, if you know of a better way please guide me, i managed to detect touch but weirdly not release of touch of the electrodes i'll now try to apply that to the original code!

gibranZawahra, 2019-04-14:
i managed to make it work,, you have to specify the i2c bus pins right before the begin statement, i used the push and release functions to programmatically push and release buttons, you'll probably find a more efficient and more stable way, here is the code iused which very simple but enough for testing and debuging, it seems to be working correctly, i saw that there 6 suggested buttons but only 4 created in the displayUI.h and only 4 updated in displayUI.cpp, why is there such a difference and are the left and right buttons declared and used used somewhere else because i can't find anything other than the pin declaration.
if  you need any other info other than this simple code please tell me i'd be happy to give back a fraction of what you have put into this.
this is the loop i only added the call for the function i added in displayUI.cpp
```
void loop() {
    currentTime = millis();

    led.update();    // update LED color
    wifiUpdate();    // manage access point
    attack.update(); // run attacks
    displayUI.update();
    displayUI.updateTouch();
    cli.update();    // read and run serial input
    scan.update();   // run scan
    ssids.update();  // run random mode, if enabled

    // auto-save
    if (settings.getAutosave() && (currentTime - autosaveTime > settings.getAutosaveTime())) {
        autosaveTime = currentTime;
        names.save(false);
        ssids.save(false);
        settings.save(false);
    }

    if (!booted) {
        // reset boot counter
        EEPROM.write(0, 0);
        EEPROM.commit();
        booted = true;
#ifdef HIGHLIGHT_LED
        displayUI.setupLED();
#endif // ifdef HIGHLIGHT_LED
    }
}
```
this is the part of the displayUI.cpp that i modified the rest is untouched:
```
#include "DisplayUI.h"
#include <Wire.h>
#include "Adafruit_MPR121.h"

Adafruit_MPR121 cap = Adafruit_MPR121();

// ===== adjustable ===== //
void DisplayUI::configInit() {

  // initialize display
  Wire.pins(5,4);
  if (!cap.begin(0x5A)) {
    Serial.println("MPR121 not found, check wiring?");
    while (1);
  }
  display.init();
....
void DisplayUI::update() {
  if (!enabled) return;

  up->update();
  down->update();
  a->update();
  b->update();

  draw();

  uint32_t timeout = settings.getDisplayTimeout() * 1000;

  if (currentTime > timeout) {
    if (!tempOff) {
      if (buttonTime < currentTime - timeout) off();
    } else {
      if (buttonTime > currentTime - timeout) on();
    }
  }
}
uint16_t lasttouched = 0;
  uint16_t currtouched = 0;
void DisplayUI::updateTouch()
{

  currtouched = cap.touched();

  for (uint8_t i = 0; i < 12; i++) {
    // it if *is* touched and *wasnt* touched before, alert!
    if ((currtouched & _BV(i)) && !(lasttouched & _BV(i)) ) {
      Serial.print(i); Serial.println(" touched");
      if(i==0){
        up->push();
      }
        if(i==1){
        down->push();
      }
        if(i==2){
        a->push();
      }
        if(i==3){
        b->push();
      }
      
    }
    // if it *was* touched and now *isnt*, alert!
    if (!(currtouched & _BV(i)) && (lasttouched & _BV(i)) ) {
      Serial.print(i); Serial.println(" released");
       if(i==0){
        up->release();
      }
       if(i==1){
        down->release();
      }
       if(i==2){
        a->release();
      }
       if(i==3){
        b->release();
      }
    }
    
  }

  // reset our state
  lasttouched = currtouched;

}
...
```
and i also added the definition for my function in the displayUI.h
```
....
   void setup();
#ifdef HIGHLIGHT_LED
    void setupLED();
#endif // ifdef HIGHLIGHT_LED
    void updateTouch();
    void update();
    void on();
    void off();

  private:
.....
```

closed 2020-09-21 by stale[bot]