SpacehuhnTech/esp8266_deauther #1087: Button Issues

Xen-Corp, 2019-04-14:
So i downloaded a copy of your source code, to flash it to my custom wemos ESP8266, but the OLED screen fails to start, so i began editing lines. One boolean was "USE_DISPLAY" and i changed it to true, and it work, until i tried using the buttons, which were extremely off, so i began to find the button-to-pin setting, in A-CONFIG.h and after editing them, nothing. In fact, it actually screwed up the OLED, to the point where i would need to fully clear the flash, and reflash a compiled binary to my ESP to get it to work. My question is, how do I change the button diagram, and is it possible to create a release that fully supports this board? "https://www.ebay.com/i/153292235548?chn=ps"

Esp8266Alfa, 2019-04-19:
https://www.hackster.io/234v244/deauthentication-attack-with-esp8266-4ed3d3

Xen-Corp, 2019-04-23:
I think i didnt mention this but, the .bin file works fine, its only the source code that screws it up. im looking for a FIXED source code

tobozo, 2019-10-09:
Closing this issue as inactive, feel free to reopen if necessary.

closed 2019-10-09 by tobozo

ryuk171, 2020-06-08:
> I think i didnt mention this but, the .bin file works fine, its only the source code that screws it up. im looking for a FIXED source code

did you find any fixed source code ??

ryuk171, 2020-06-09:
i tried it with another versions of source code it works fine , as mr.kremser said the new version might be buggy !!