SpacehuhnTech/esp8266_deauther #1088: No Deauth packets

cab19, 2019-04-18:
Hi,

I've had a hard time getting this to flash on a NodeMCU ver 3 (yeah the big ugly one). I followed the instructions. I tried flashing the latest bin file with ESP8266Flasher and got fatal exceptions.

I then tried to upload the **blink** or **wifiscan** examples
- via Arduino using ESP8266 Deauther Modules ( NodeMCU 1.0 ) = fatal exceptions
- Upload via Arduino using ESP8266 Boards (v2.0)  ( NodeMCU 1.0 ) = fatal exceptions

I finally found that using ESP8266 Boards (v2.5) I could flash examples to the board and it would work.
So I ended up having to comment out all the Display related code to get the project to compile, also added ArduinoJson 5.13.5. and it finally uploaded. I can access the interface, I can change and save settings, I can do a beacon attack but trying a deauth attack never works. As you can see below, from my snipped console, deauth never sends any packets, although beacon attack does.

Any ideas please!

```
21:44:25.742 -> [Pkt/s] All:    0 | Deauths:   0/50  | Beacons:   0/0   | Probes:   0/0 

21:44:46.385 -> [Pkt/s] All:   80 | Deauths:   0/0   | Beacons:  80/80  | Probes:   0/0  
21:44:47.398 -> [Pkt/s] All:   64 | Deauths:   0/0   | Beacons:  64/80  | Probes:   0/0  
21:44:48.414 -> [Pkt/s] All:   80 | Deauths:   0/0   | Beacons:  80/80  | Probes:   0/0

cab19, 2019-04-19:
Oh Probes works too, it's only Deauths never sending any packets.

sro2, 2019-04-19:
What is the probrequest attack can you explain plz ..... I dont khonw how
it works

cab19, 2019-04-20:
I believe the probe is sent to the AP requesting information about the AP, not sure what the purpose is. My main interest was in the deauth. I thought I'd just mention that the log captures data for the three and how many packets are sent. Beacons and Probes send packets, Deauth never does. :\

tobozo, 2019-10-09:
Closing this issue as inactive, feel free to reopen if necessary.

closed 2019-10-09 by tobozo

ngtrung1992, 2020-06-14:
reopen my case 
reproduce :     
-I select one or multi APs 
-On attack tab 
-click start 

( deauth method can not send any ptks but after symbol slash have number (Deauths:   0/40)
probes and beacons method still work )

[Pkt/s] All:    0 | Deauths:   0/40  | Beacons:   0/0   | Probes:   0/0 
[Pkt/s] All:    0 | Deauths:   0/40  | Beacons:   0/0   | Probes:   0/0 
[Pkt/s] All:    0 | Deauths:   0/40  | Beacons:   0/0   | Probes:   0/0 
[Pkt/s] All:    0 | Deauths:   0/40  | Beacons:   0/0   | Probes:   0/0 
[Pkt/s] All:    0 | Deauths:   0/40  | Beacons:   0/0   | Probes:   0/0 

Issue when flash with my arduino 1.8.12 ( i just config settings.h file  & installed board 2.1.0 ). no error show in when upload project to module.

Please reopen this case and tell me anything to solve.
p/s: No issue if auto flash with n2d2 tool

spacehuhn, 2020-06-16:
>  installed board 2.1.0
You need 2.0.0 or better: the Deauther SDK by Tobozo
https://github.com/SpacehuhnTech/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide