SpacehuhnTech/esp8266_deauther #1090: Problem with modified Web Interface

SentinelMKX, 2019-04-19:
<h1>Problem with modifed Web Interface:</h1>
<hr />
<h2>What I done so far:</h2>
<strong> 
<ol>
<li>succesfully uploaded deauther 2.0 sketch</li>
<li>changed all html, js files I modified in archives ".gz"</li>
<li>succesfully uploaded sketch "data" via ESP8226FS plugin</li>
<li>changed "WEBSPIFFS" value to "true", everywhere in code also in web interface</li>
<li>reboot ESP8266 multiple times</li>
</ol>
<hr />
</strong>
So, my problem is that I can't get my custom Web Interface working, I done _things above_ and I still can't get it working, and I have completely no idea why.  
Yeah, I don't know what else to say or what other information provide, there is not much.
But if you will tell me, I will try to provide all the needed information. but I don't even know what information would be useful now. 
On wiki there is not much about it, and I didn't find any solution on internet.

SentinelMKX, 2019-04-19:
Okay, for unknown reasons it started working.
I used the reset_4mb.bin to clean it and I flash it again same way and it worked so yeah, I apologize

closed 2019-04-19 by SentinelMKX

garustea, 2020-10-23:
@Arisa-Snowbell  Can I ask you step by step which procedure you used to modify the html? I've been trying for two days with no results. Thanks in advance

spacehuhn, 2020-10-26:
The procedure to is very easy as long as you only modify existing files:
1. Make your changes to the files in the folder [`web_interface`](https://github.com/SpacehuhnTech/esp8266_deauther/tree/v2/web_interface)
2. Run [`utils/webconverter.py`](https://github.com/SpacehuhnTech/esp8266_deauther/tree/v2/utils/web_converter) 
3. Upload the code using Arduino

If you add or remove files, you'll need to edit [`wifi.h`](https://github.com/SpacehuhnTech/esp8266_deauther/blob/v2/esp8266_deauther/wifi.h#L233) as well.

If you make changes and don't see the result, try clearing the browser cache.