SpacehuhnTech/esp8266_deauther #1093: Undefined reference to 'sscanf'

ToxicCow19, 2019-04-29:
Hi. I tried making the project but I'm getting this compile error:

    Archiving built core (caching) in: C:\Users\alvar\AppData\Local\Temp\arduino_cache_849758\core\core_deauther_esp8266_nodemcuv2_CpuFrequency_160,UploadTool_esptool,FlashFreq_80,FlashMode_dio,UploadSpeed_115200,FlashSize_1M256_8b84c6b551e21998fdbadb8222d1dc23.a
    libraries\Arduino_JSON\cjson\cJSON.c.o:(.text.print_value+0x24): undefined reference to `sscanf'

    libraries\Arduino_JSON\cjson\cJSON.c.o: In function `print_value':

    C:\Users\alvar\Documents\Arduino\libraries\Arduino_JSON\src\cjson/cJSON.c:2964: undefined reference to `sscanf'

    collect2.exe: error: ld returned 1 exit status

    exit status 1
    Error compiling for board NodeMCU 1.0 (ESP-12E Module)

I tried unistalling and installing Arduino_JSON.h multiple times and I made sure I was compiling for the deauther.

The version of Arduino_JSON is 0.1.0 (the only one I can install)

tobozo, 2019-04-29:
when multiple libraries are detected, you can't predict which one Arduino IDE will pick

You may safely delete this folder : `C:\Users\alvar\Documents\Arduino\libraries\Arduino_JSON\` and try to compile again

ToxicCow19, 2019-04-30:
Thanks, it worked!

closed 2019-04-30 by ToxicCow19

majidshakhsi, 2020-09-14:
Hello everyone,
I have a problem with Arduino_json.h in my node_mcu project.
when i compile it, i face with a problem as follow,
C:\Users\Majid\AppData\Local\Temp\arduino_build_48742\libraries\Arduino_JSON\cjson\cJSON.c.o: In function `print_string_ptr':
C:\Users\Majid\Documents\Arduino\libraries\Arduino_JSON\src\cjson/cJSON.c:81: undefined reference to `__ctype_ptr__'
C:\Users\Majid\AppData\Local\Temp\arduino_build_48742\libraries\Arduino_JSON\cjson\cJSON.c.o:(.text.print_value+0x24): undefined reference to `sscanf'
C:\Users\Majid\AppData\Local\Temp\arduino_build_48742\libraries\Arduino_JSON\cjson\cJSON.c.o: In function `print_value':
C:\Users\Majid\Documents\Arduino\libraries\Arduino_JSON\src\cjson/cJSON.c:81: undefined reference to `sscanf'
collect2.exe: error: ld returned 1 exit status
Using library ESP8266WiFi at version 1.0 in folder: C:\Users\Majid\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi 
Using library ESP8266HTTPClient at version 1.1 in folder: C:\Users\Majid\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266HTTPClient 
Using library Arduino_JSON at version 0.1.0 in folder: C:\Users\Majid\Documents\Arduino\libraries\Arduino_JSON 
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

I delete Arduino_json.h from C:\Users\Majid\Documents\Arduino\libraries\Arduino_JSON\  and then I install it again.
But, unfortunately, I have this error in the project.
Please guide me what happen by such an error.
Thanks

Bommlon, 2020-09-16:
@majidshakhsi please open a new issue for your problem. Make sure to go through the template provided when opening one.