SpacehuhnTech/esp8266_deauther #1096: Deauth wont work

MadMax1611, 2019-05-14:
Hey,

So i got a ESP8266 Board and flashed it with the 1mb Deauther Package. The Interface and Stuff works fine but Deauth wont work. I tried it with my Router and Mobile Hotpsot. Nothing worked. Is there any fix?

jLynx, 2019-05-14:
And you can confirm these are 2.4 devices?

MadMax1611, 2019-05-14:
Not really. So my Fritzbox is using Hybrid 2,4 and 5 Ghz at the Same Time. But my Mobile Hotspot should be 2.4 Ghz. Hotspot  from Huawei P20 and the Device i connected to it is my old Xperia Z3. That should use 2.4

jLynx, 2019-05-14:
Issue could quite possibly be that they are connecting to 5ghz networks which means it wont do anything. Try turn off the hybrid mode and only turn on 2.4 or just make a dedicated 2.4 mode to test on

MadMax1611, 2019-05-14:
Ok so i forced my Mobile Hotspot now to use 2.4 Ghz and now its working. Thanks for the Tip. So it just works with 2.4 Ghz Devices? No Fix for 5 Ghz Devices?

jLynx, 2019-05-14:
It's all in the FAQ. There is no "fix" for 5ghz as this device was never supposed to work with them to begin with as specified.

closed 2019-05-14 by jLynx