SpacehuhnTech/esp8266_deauther #1097: Exception(0); if using Emergency-Deauth-Button-Script

Cherubrine, 2019-05-14:
I have tried uploading the "Deauth All Emergency Button Script" to my NodeMCU and the first time it worked perfectly.
But I wanted to rename the default AP name, so I uploaded the resetScript and reuploaded the changed deauther script from Arduino, but it is not working, all LED's are blinking at the same time but no AP is opening, I just get an error in the CLI.

I have tried with a portable version of Arduino, restarted my computer, flashed the blank .bin
and tried it with 3 different NodeMCU boards, but it won't work.<br>
It works if I upload the default script.<br>
 The only code I have added is: 
```
if(!attack.isRunning()){
      cli.runCommand("stopap");               // stop access point and web interface
      cli.runCommand("scan aps -c 60s");      // start scan for access points each minute
      cli.runCommand("attack -da");           // start deauth all
}
```
I added it directly after the "void loop() {" , and before the "currentTime = millis();"

It would be nice if someone could help me, I don't know any other solution!<br>
<details>
  <summary> Compile log (...)</summary>

  Switched to Channel 9
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '9', hidden: false, captive-portal: true
ST
Exception (0):
epc1=0x4022a0fc epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff1870 end: 3fff1df0 offset: 01a0

>>>stack>>>
3fff1a10:  00000001 00000000 00000002 40104023  
3fff1a20:  40103894 3ffedcc0 3fffc250 4000050c  
3fff1a30:  400043da 00000030 00000018 ffffffff  
3fff1a40:  400044ab 3fffc718 3fff1b30 08000000  
3fff1a50:  60000200 08000000 00000003 00000000  
3fff1a60:  0000ffff 00000001 04000002 000fd000  
3fff1a70:  3fff5150 000000fd 3ffef294 00000030  
3fff1a80:  00000002 00000000 00000010 3fffd85c  
3fff1a90:  000ed000 000ed100 000ed100 00000030  
3fff1aa0:  00000014 400042db 40104f68 3fff1b10  
3fff1ab0:  3fff1b10 40233b3e 3fff6490 3fff2888  
3fff1ac0:  3ffe9fe0 3fff1bf0 3fff2858 ffffffff  
3fff1ad0:  40233b67 000bc42c 00000004 3fff1af0  
3fff1ae0:  00000002 000bc400 3fff1b10 3fff1b10  
3fff1af0:  00000008 0000000f 00000000 0000002a  
3fff1b00:  3fff1b10 3fff1b10 00000008 40235c52  
3fff1b10:  00000010 40235c3d 000000ad 00000076  
3fff1b20:  3fff2928 4022c338 3fff2404 3fff2888  
3fff1b30:  00000002 4000410f 3fff2888 4022c4a6  
3fff1b40:  3fffc718 40004a3c 000000fd 3fff5150  
3fff1b50:  3fffc718 401019c0 00000378 40233e41  
3fff1b60:  000000fd 40235223 00000378 000000ff  
3fff1b70:  00001000 402352b9 3ffef294 000000ff  
3fff1b80:  000000fd 3fff1bf0 000fd000 3ffef294  
3fff1b90:  ffffff00 55aa55aa 00000020 0000001c  
3fff1ba0:  0000001c 000000ee 000000ed 000000ff  
3fff1bb0:  40235160 00000001 3ffefe0a 3fff0bd4  
3fff1bc0:  00000001 4023518e 00000001 4020957a  
3fff1bd0:  4021c8eb 3ffefe0a 00000001 4021c8e6  
3fff1be0:  00000000 40256258 3fff0920 4020c335  
3fff1bf0:  0000000f 00000000 0000000f 00000002  
3fff1c00:  3fff2ec0 0000000f 00000006 3ffe8c00  
3fff1c10:  3fff1c70 3fff1c90 3fff1ef0 4020517a  
3fff1c20:  3fff2eb0 4010695e 3fff1cd0 00000001  
3fff1c30:  0000000e 3fff0001 00000004 4022ad2a  
3fff1c40:  3fff2eb0 4010695e 3fff1cf0 00000004  
3fff1c50:  40226038 3fff1ca0 3fff1cd0 4022ad2a  
3fff1c60:  3fff1cb0 00000004 3fff1cf0 4022ad82  
3fff1c70:  3fff0d04 00000001 3ffe8aa5 4022a310  
3fff1c80:  40267810 7fffffff 3fff1cf0 3fff2ec0  
3fff1c90:  3ffe8c10 00000004 3fff0d04 40225f1d  
3fff1ca0:  3ffe8aa4 00000011 3fff1cf0 3fff2ec0  
3fff1cb0:  3ffe8c10 00000000 3fff0d04 4022a3e4  
3fff1cc0:  40106970 3ffefe74 3fff2ec0 0000000f  
3fff1cd0:  00000006 00000000 3ffefe74 4022acdc  
3fff1ce0:  3ffe8c10 00000000 3ffefe74 4021d02c  
3fff1cf0:  3fff2ec0 0000000f 00000004 4021c187  
3fff1d00:  3ffe8c10 00000000 3ffefe74 4021d823  
3fff1d10:  3fff2ec0 0000000f 4021bac0 4021eff4  
3fff1d20:  3fff2ec0 0000000a 3fff0d04 40225a75  
3fff1d30:  3fff0d04 3fff2998 0000000f 00000006  
3fff1d40:  3fff2968 4010695e 3fff1db0 00000006  
3fff1d50:  3fff1db0 00000006 3fff0d04 4022ad2a  
3fff1d60:  3ffe8c14 00000006 3fff1db0 4022ad82  
3fff1d70:  3fff0920 3fff0d04 00000000 00000005  
3fff1d80:  00000000 00000000 3fff1db0 4022ae00  
3fff1d90:  3ffe8c14 3fff0d04 3fff0b58 3fff0dc8  
3fff1da0:  3fff0944 00000000 3fff0920 4021d9b2  
3fff1db0:  3fff2978 0000000f 00000006 00000000  
3fff1dc0:  00000000 00000000 00000000 3fff0dc8  
3fff1dd0:  3fffdc20 00000000 3fff0dc1 40226149  
3fff1de0:  00000000 00000000 3fff0dd0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Mounting SPIFFS...OK
Formatting SPIFFS...OK
Switched to Channel 9
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '9', hidden: false, captive-portal: true
ST
Exception (0):
epc1=0x4022a0fc epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff1870 end: 3fff1df0 offset: 01a0

>>>stack>>>
3fff1a10:  4010524f 00080000 3fff1a50 4024faf2  
3fff1a20:  00000000 00000000 3fffc250 4000050c  
3fff1a30:  400043e6 00000030 00000018 ffffffff  
3fff1a40:  400044ab 3fffc718 3fff1b30 08000000  
3fff1a50:  60000200 08000000 08000000 00000000  
3fff1a60:  0000ffff 00000001 04000002 000fe000  
3fff1a70:  3fff5138 000000fe 3ffef294 00000030  
3fff1a80:  00000002 00000000 00000010 3fffd85c  
3fff1a90:  00000008 00000008 00000000 3fff1ef0  
3fff1aa0:  00000014 400042db 00000008 3fff1b10  
3fff1ab0:  3fff1b10 40233b3e 3fff6478 3fff2888  
3fff1ac0:  3ffe9fe0 3fff1bf0 3fff2858 40226174  
3fff1ad0:  40233b67 000bc42c 00000004 3fff1af0  
3fff1ae0:  00000002 000bc400 3fff1b10 3fff1b10  
3fff1af0:  00000008 00000010 40248fd9 3ffedcc0  
3fff1b00:  3fff1b10 3fff1b10 00000008 40235c52  
3fff1b10:  00000010 40235c3d 000000ad 00000076  
3fff1b20:  3fff2928 4022c338 3fff2404 3fff2888  
3fff1b30:  00000002 4000410f 3fff2888 4022c4a6  
3fff1b40:  3fffc718 40004a3c 000000fe 3fff5138  
3fff1b50:  3fffc718 401019c0 00000378 40233e41  
3fff1b60:  000000fe 40235223 00000378 000000ff  
3fff1b70:  00001000 402352b9 3ffef294 000000ff  
3fff1b80:  000000fd 3fff1bf0 000fe000 3ffef294  
3fff1b90:  ffffff01 55aa55aa 00000023 0000001c  
3fff1ba0:  0000001c 000000ed 000000ee 000000ff  
3fff1bb0:  40235160 00000001 3ffefe0a 3fff0bd4  
3fff1bc0:  00000001 4023518e 00000001 4020957a  
3fff1bd0:  4021c8eb 3ffefe0a 00000001 4021c8e6  
3fff1be0:  00000000 40256258 3fff0920 4020c335  
3fff1bf0:  0000000f 00000000 0000000f 00000002  
3fff1c00:  3fff2ec0 0000000f 00000006 3ffe8c00  
3fff1c10:  3fff1c70 3fff1c90 3fff1ef0 4020517a  
3fff1c20:  3fff2eb0 4010695e 3fff1cd0 00000001  
3fff1c30:  0000000e 3fff0001 00000004 4022ad2a  
3fff1c40:  3fff2eb0 4010695e 3fff1cf0 00000004  
3fff1c50:  40226038 3fff1ca0 3fff1cd0 4022ad2a  
3fff1c60:  3fff1cb0 00000004 3fff1cf0 4022ad82  
3fff1c70:  3fff0d04 00000001 3ffe8aa5 4022a310  
3fff1c80:  40267810 7fffffff 3fff1cf0 3fff2ec0  
3fff1c90:  3ffe8c10 00000004 3fff0d04 40225f1d  
3fff1ca0:  3ffe8aa4 00000011 3fff1cf0 3fff2ec0  
3fff1cb0:  3ffe8c10 00000000 3fff0d04 4022a3e4  
3fff1cc0:  40106970 3ffefe74 3fff2ec0 0000000f  
3fff1cd0:  00000006 00000000 3ffefe74 4022acdc  
3fff1ce0:  3ffe8c10 00000000 3ffefe74 4021d02c  
3fff1cf0:  3fff2ec0 0000000f 00000004 4021c187  
3fff1d00:  3ffe8c10 00000000 3ffefe74 4021d823  
3fff1d10:  3fff2ec0 0000000f 4021bac0 4021eff4  
3fff1d20:  3fff2ec0 0000000a 3fff0d04 40225a75  
3fff1d30:  3fff0d04 3fff2998 0000000f 00000006  
3fff1d40:  3fff2968 4010695e 3fff1db0 00000006  
3fff1d50:  3fff1db0 00000006 3fff0d04 4022ad2a  
3fff1d60:  3ffe8c14 00000006 3fff1db0 4022ad82  
3fff1d70:  3fff0920 3fff0d04 00000000 00000005  
3fff1d80:  00000000 00000000 3fff1db0 4022ae00  
3fff1d90:  3ffe8c14 3fff0d04 3fff0b58 3fff0dc8  
3fff1da0:  3fff0944 00000000 3fff0920 4021d9b2  
3fff1db0:  3fff2978 0000000f 00000006 00000000  
3fff1dc0:  00000000 00000000 00000000 3fff0dc8  
3fff1dd0:  3fffdc20 00000000 3fff0dc1 40226149  
3fff1de0:  00000000 00000000 3fff0dd0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Mounting SPIFFS...OK
Formatting SPIFFS...

</details>

> 

> 

> 

> 

> 

>

Cherubrine, 2019-05-14:
Ok, I've uploaded it 5 times in a row and it worked.
I don't know the reason of the error, but as long as it works, I don't care.

Kyuugeki, 2019-05-22:
I'm having the same problem and same error :(
Reuploading didn't fix it 

What should i do?

(I'm uploading it to a ESP12E module, But im having the same issue with a NodeMCU 1.0).

Cherubrine, 2019-05-23:
I just uploaded the reset file and then uploaded the deauther file 5 times and it worked.

closed 2019-10-09 by tobozo