SpacehuhnTech/esp8266_deauther #1099: Please Update your Bin files the last you update was at November

JadenJSJ, 2019-05-23:
I use the deauthing program and tried deauthing my friends laptop is using a TP-link wifi usb adapter
 The laptop didnt get kicked of the wifi and i tried it on other devices and they get kicked of the wifi is there any solution to this problem my wifi router is 2.4 ghz 

[image]

Wifi Usb Name is AC600 Archer T2UH High Gain TP-link Image:

 [image]

I want use it for pranking my friend because he use a airhorn on me at 4:00 AM and i cant wait to see the next update can you tell me when the next update will be and also for the update 3.0 and a pull request is 
*Rounded icons and buttons
*Better Settings
*Option to lower the signal that brodcast the control panel website
*Support for Windows (if you don't understand i mean please open a new section named windows_deauthur so we can deauth in our computers instead of doing them on a esp8266)
*And support for time like (set it to deauthur this device a for 3 mins and set to deauthur at 3:00 PM
*Can you please add a dev and beta option if possible
*thanks for making this project
I hope the fix and features will be solved faster and added faster

> Thank You For Reading

M0ddedNiNja, 2019-05-23:
I think I can solve your problems
1. I don't think you can deauth your friend because his adapter uses the 802.11n (i think) standard which can protect its self from deauth attacks
2. You can do these attacks from windows by buying a supported wifi adapter and installing aircrack-ng

JadenJSJ, 2019-05-24:
Is there a way around to do this?

closed 2019-10-09 by tobozo