SpacehuhnTech/esp8266_deauther #1101: expected primary-expression before 'display'

tvmp, 2019-05-25:
Hello and thank you for your project.

While compiling with Arduino IDE, I am getting the following error:

---------------------------------------

In file included from sketch\DisplayUI.h:5:0,

                 from sketch\DisplayUI.cpp:1:

A_config.h:77:58: error: expected primary-expression before 'display'

 #define DEAUTHER_DISPLAY SSD1306Spi display = SSD1306Spi display = SSD1306Spi(D1, D2, D8); // for 0.96" OLED with SPI

                                                          ^

sketch\DisplayUI.h:80:9: note: in expansion of macro 'DEAUTHER_DISPLAY'

         DEAUTHER_DISPLAY // see config.h

         ^

In file included from sketch\CLI.h:11:0,

                 from sketch\CLI.cpp:1:

A_config.h:77:58: error: expected primary-expression before 'display'

 #define DEAUTHER_DISPLAY SSD1306Spi display = SSD1306Spi display = SSD1306Spi(D1, D2, D8); // for 0.96" OLED with SPI

                                                          ^

sketch\DisplayUI.h:80:9: note: in expansion of macro 'DEAUTHER_DISPLAY'

         DEAUTHER_DISPLAY // see config.h

         ^

Multiple libraries were found for "ArduinoJson.h"
 Used: D:\BackUp\ARDUINO\SKETCHBOOK\libraries\ArduinoJson
 Not used: C:\Users\Alexander\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\ArduinoJson
exit status 1
In file included from sketch\Settings.h:11:0,

                 from D:\BackUp\ARDUINO\PROJECTS\DEAUTHER\esp8266_deauther-2.1.0\esp8266_deauther\esp8266_deauther.ino:27:

A_config.h:77:58: error: expected primary-expression before 'display'

 #define DEAUTHER_DISPLAY SSD1306Spi display = SSD1306Spi display = SSD1306Spi(D1, D2, D8); // for 0.96" OLED with SPI

                                                          ^

sketch\DisplayUI.h:80:9: note: in expansion of macro 'DEAUTHER_DISPLAY'

         DEAUTHER_DISPLAY // see config.h

         ^

expected primary-expression before 'display'

---------------------------------------

I am using an OLED SSD1306 SPI.

Could you please help me trace what I am doing wrong?

tobozo, 2019-05-25:
looks like you have duplicate libraries in your BACKUP folder (i.e delete the folder `D:\BackUp\ARDUINO\SKETCHBOOK\libraries\ArduinoJson` and at least one future error will go away :smile: 

Also you copied twice the value in your editing of the config file :

`#define DEAUTHER_DISPLAY SSD1306Spi display = SSD1306Spi display = SSD1306Spi(D1, D2, D8);`

should be 

`#define DEAUTHER_DISPLAY SSD1306Spi display = SSD1306Spi(D1, D2, D8);`

happy hacking ^^

tvmp, 2019-05-25:
Thank you tobozo, it worked perfectly!

closed 2019-05-25 by tvmp