SpacehuhnTech/esp8266_deauther #1105: No network found

Danielr19, 2019-05-29:
//Found an issue with a similar problem but found no solutions

Hi, I really liked your project and I think that what you are doing is great and spiring, at least for me. I tried installing the 2.0.5 version on a board I´ve got, everything seemed fine, I was even able to upload the code. The wifi network "pwned" won't appear, settings are as shown in the first picture. I'll try with another version but I wanted to know if the problem is due to the version I used. I'll post some pictures showing the board I own.

[image]

[image]

[image]

Danielr19, 2019-05-29:
Update: Tried it with the easter version and I had the same problem

SentinelMKX, 2019-05-29:
umm, I think you have wrong development board, I don't see network chip on your board

jLynx, 2019-05-29:
@SentinelMKX, I do. It's the bottom chip

Danielr19, 2019-05-30:
It has a wifi chip, I've used this board in other projects small projects involving wifi, but I don't know why I cant get it working for this one.
btw thx for trying to help, I really want to able to show this project to my friends.

closed 2019-10-09 by tobozo