SpacehuhnTech/esp8266_deauther #1108: change Radio Type from 802.11g to 802.11n

RvdHout, 2019-06-04:
I am very new to this, i therefor simply installed the deauther_2.1.0_1mb.bin on my ESP8266 and did not try to compile it myself.
The chip on the ESP8266 says it supports 802.11b/g/n

Both the 'pwned' SSID or any other SSID created as beacons show up as 802.11g WiFi networks, is there a way to make all of them show up as 802.11n WiFi networks?

RvdHout, 2019-06-05:
I have read that possibly i could switch from 802.11g to 802.11n using Arduino IDE, something like this:

`if(wifi_get_phy_mode() != PHY_MODE_11N)
wifi_set_phy_mode(PHY_MODE_11N);`

Anyone knows where to do this in the esp8266_deauther code?

spacehuhn, 2019-06-05:
I don't think that's the issue here.
You'd need to append some info to the beacon frame to be recognized as a N network.
The byte array should be in Attack.h and here you can find more info about the beacon frames: https://mrncciew.com/2014/10/08/802-11-mgmt-beacon-frame/

RvdHout, 2019-06-05:
Uh...didn't i mention i am very new to still all :)
So, it is by design deauther AP mode as beacons are identified as 802.11g WiFi networks?

Can you give some hints on the byte array that should be in Attack.h?

spacehuhn, 2019-06-05:
https://github.com/spacehuhn/esp8266_deauther/blob/master/esp8266_deauther/Attack.h#L155

spacehuhn, 2019-06-05:
The packet probably needs another tag to specify the n compatibility and the supported rates.

RvdHout, 2019-06-05:
I can't get this figured out...to bad!

RvdHout, 2019-06-06:
Even the above linked 'wifijam ‒ An 802.11n beacon frame transmission program' broadcast as being 802.11g, wtf? Maybe my ESP8266 is lacking 802.11n? 
....it was a really cheap Chinese eBay purchase, although the chip on the ESP8266 says it supports 802.11b/g/n

RvdHout, 2019-06-09:
I managed to change te radio type for beacons, it did cost me a whole weekend though :)

closed 2019-06-09 by RvdHout