SpacehuhnTech/esp8266_deauther #1109: Arduino Sources error

bog244, 2019-06-05:
doesnt loading modules.. i just got some errors, when going to ManagerBoard.. with another sources it worked.. but with this :/

http://arduino.esp8266.com/stable/package_esp8266com_index.json
http://phpsecu.re/esp8266/package_deauther_index.json

tobozo, 2019-06-05:
"some errors" isn't enough to get help, please edit your post and add more details

mischk, 2019-07-03:
Cant install `arduino-esp-8266-deauther`:

` Error downloading http://arduino.esp8266.com/linux64-xtensa-lx106-elf-gb404fb9.tar.gz
java.lang.RuntimeException: java.lang.Exception: Error downloading http://arduino.esp8266.com/linux64-xtensa-lx106-elf-gb404fb9.tar.gz
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:176)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading http://arduino.esp8266.com/linux64-xtensa-lx106-elf-gb404fb9.tar.gz
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:131)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:69)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:110)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:173)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 404
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:194)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:129)
	... 4 more`

there seems to be a problem with the sources.

tobozo, 2019-07-03:
looks like the 404 problem is coming from the [now obsolete json file](http://arduino.esp8266.com/stable/package_esp8266com_index.json)

possible fix by using the release URL (e,g, [this one](https://github.com/esp8266/Arduino/releases/download/2.5.2/package_esp8266com_index.json)) ?

mischk, 2019-07-04:
no, the error happens when trying to install the `arduino-esp-8266-deauther` Changing the 'package_esp8266com_index.json'- URL does not work. Im using Linux and Arduino 1.8.9

tobozo, 2019-07-04:
The suggestion I gave you isn't based on beliefs but on facts: I'm also using Arduino 1.8.9 and the whole removal / reinstall process worked just fine by using this esp8266 package URL:

    https://github.com/esp8266/Arduino/releases/download/2.5.2/package_esp8266com_index.json

instead of this **now obsolete** esp8266 package URL:

    http://arduino.esp8266.com/stable/package_esp8266com_index.json

Before doing that: same symptoms as those you describe
After doing that: fixed, no more error, compilation works (haven't tested the deauther code though)

As far as the other deauther package is concerned I even went to check all URLs (e.g. containing the string "gb404fb9") since it was recently maintained, but everything was fine, all links responded with a file to download.

    https://phpsecu.re/esp8266/package_deauther_index.json

[image]

If your Arduino installation is buggy after using wrong URLs, maybe a cleanup is required?

closed 2019-10-09 by tobozo