SpacehuhnTech/esp8266_deauther #1112: Deauther 2.1.0 on WeMos D1 Mini(ESP8266) - Battery Level

Ch4ist0, 2019-06-13:
Hello, I am using the WeMos D1 mini system (ESP8266) is it possible to change the battery level options in the menu Deauther 2.1.0 instead of "clock"?

TheMiguelBi, 2019-12-11:
How did you manage to run version 2.1 on wemos d1 mini ?? O.O

spacehuhn, 2020-06-16:
>  change the battery level options
Not sure I fully understand your question, but you need extra hardware to read the battery voltage. Such a feature is not implemented so you'd have to code it yourself.

GrandJohnny, 2020-09-16:
> 
> 
> > change the battery level options
> 
> Not sure I fully understand your question, but you need extra hardware to read the battery voltage. Such a feature is not implemented so you'd have to code it yourself.

hello there
how can we define if there is more menu nodes so it goes to next page?
for example if i add a new option on the menu so they become 6 options but only 5 can be displayed (while my oled is big enough and configured if there would be more content ) ?

Bommlon, 2020-09-24:
@GrandJohnny I'm not sure how this is connected to this issue. Please open a new one if you don't mind. It would make it easier for other people with the same question to find it.

GrandJohnny, 2020-09-24:
> 
> 
> @GrandJohnny I'm not sure how this is connected to this issue. Please open a new one if you don't mind. It would make it easier for other people with the same question to find it.

#1314

closed 2020-12-30 by stale[bot]