SpacehuhnTech/esp8266_deauther #1125: Feature Request: Randomize Deauthentication Reason Code

projectb-temp, 2019-07-23:
Hi there and thank you for this great project you keep maintaining for so long.

Would it be possible to extend the `DEAUTH-REASON` setting so it is possible to not only use a fixed reason code but also to randomize the Deauth Reason Code (or at least have it switch between some valid/usual codes) so it does not always use the same code again and again?

Thank you a lot and keep up the good work.

Esp8266Alfa, 2019-07-25:
How do i check the reason code is sent exactly as changed in the web interface?
I edited the source code so I can change the reason code in the serial interface.  From there I wrote a script to change the reason code by the time specified and repeat it.  But I don't know what it's sending.
sorry for my english not being good

tobozo, 2019-07-25:
@Esp8266Alfa see [Wireshark](https://github.com/wireshark/wireshark)