SpacehuhnTech/esp8266_deauther #1126: Cant use anything anymore!

getPoland, 2019-07-24:
> I cant flash anymore on my nodes... i after this error i tried to flash blink on the node but it gave me the same error so i tried another node but it also does not work...
I hope you can help me!

<details>
  <summary> Compile log (...)</summary>

In file included from sketch\Attack.h:9:0,

                 from sketch\Attack.cpp:1:

C:\Users\rzepk\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include/user_interface.h:451:39: error: 'freedom_outside_cb_t' was not declared in this scope

 int wifi_register_send_pkt_freedom_cb(freedom_outside_cb_t cb); 

                                       ^

In file included from sketch\ClientScan.h:13:0,

                 from sketch\ClientScan.cpp:1:

C:\Users\rzepk\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include/user_interface.h:451:39: error: 'freedom_outside_cb_t' was not declared in this scope

 int wifi_register_send_pkt_freedom_cb(freedom_outside_cb_t cb); 

                                       ^

Mehrere Bibliotheken wurden für "WiFiUdp.h" gefunden
 Benutzt: C:\Program Files (x86)\Arduino\libraries\WiFi
 Nicht benutzt: C:\Users\rzepk\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi
In file included from C:\Users\rzepk\OneDrive\Desktop\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino:54:0:

C:\Users\rzepk\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include/user_interface.h:451:39: error: 'freedom_outside_cb_t' was not declared in this scope

 int wifi_register_send_pkt_freedom_cb(freedom_outside_cb_t cb); 

                                       ^

exit status 1
Fehler beim Kompilieren für das Board Generic ESP8266 Module.
</details>
<br><br><br>

13r1ckz, 2019-10-08:
first try to upload a blinky script. if this works follow this link: https://github.com/spacehuhn/esp8266_deauther/wiki/Installation#flashing-the-firmware-bin-file

when the blinky script does not work there are two options left. your esp module is dead, the other option is that u use the wrong blinky script.

closed 2019-10-08 by 13r1ckz

getPoland, 2020-01-02:
Sorry, it was my issue i used an older version of the IDE and the wrong driver for the ESP module