SpacehuhnTech/esp8266_deauther #1128: What cord transfers data between the NodeMCU ESP8266 and a windows laptop?

cMik3s1gA, 2019-07-28:
I've read why no coms show up on my NodeMCU flash. I had the same problem and read that I need a different cord to transfer data. What cord do I need?

don-willingham, 2019-07-30:
Ensure you have the correct drivers installed. See https://github.com/spacehuhn/esp8266_deauther/wiki/Installation#drivers-and-com-port
Alternately, you can use a "USB to TTL", make sure it's 3.3v or set to 3.3v if it also supports 5v; to connect: transmit to receive, receive to transmit, ground to ground. I have [this one](https://www.amazon.com/JBtek-WINDOWS-Supported-Raspberry-Programming/dp/B00QT7LQ88/ref=sr_1_1?keywords=JBtek+Windows+8+Supported+Debug+Cable+for+Raspberry+Pi+USB+Programming+USB+to+TTL+Serial+Cable&qid=1564450714&s=electronics&sr=1-1)

commanderHR1, 2019-08-23:
Make sure that the cable you are using is NOT a charge-only cable. It must be able to transmit data. As don-willingham said, make sure you also have the correct drivers and settings. You could also need an OTG (On the Go) adapter for your cable as well. 
Hope this helps.

closed 2019-10-09 by tobozo