SpacehuhnTech/esp8266_deauther #1130: Temperature sensors

chiragvshroff, 2019-07-29:
Hey guys i just wanted to know if the esp8266 has a temperature sensor

don-willingham, 2019-07-30:
The esp8266 is a microcontroller. It's available on many different circuit boards. Typically a temperature sensor is connected to GPIO pins, depending on the specific sensor's interface. There could be a board available with a temperature sensor pre-installed... but what does temperature have to do with esp8266_deather?

commanderHR1, 2019-08-23:
It doesn't come with a built-in temperature sensor, as the ESP8266 isn't a board at all. It's a microcontroller (think like a CPU, but semi-different). There are boards that INCLUDE the ESP8266, but the ESP8266 itself is only a component. To answer your question, the actual microcontroller does not, however, there are boards that have the ESP8266, WiFi capabilities and can interface with sensors. I would recommend the [Adafruit Feather Huzzah ESP8266 board](https://www.adafruit.com/product/3046), and you can use an I2C temp sensor with it. 

TL;DR No, the microcontroller doesn't, but there are boards that can work with a temperature sensor (or have them built-in).

tobozo, 2019-10-09:
Closing this issue as out of scope, feel free to reopen if necessary.

closed 2019-10-09 by tobozo