SpacehuhnTech/esp8266_deauther #1132: Clone SSID creates 60

CryptoCubik, 2019-08-01:
So I'm messing around with my new device and I'm not sure what all the options do.
I noticed I can clone SSIDs, but I don't know why that would be useful, nor do I know why cloning an SSID creates 60(!!!) copies of it instead of just 1.
Why?

projectb-temp, 2019-08-01:
Then you should start reading the project's Wiki: https://github.com/spacehuhn/esp8266_deauther/wiki
And get yourself some basic knowledge about things like...
- WiFi deauthentication Attacks: https://en.wikipedia.org/wiki/Wi-Fi_deauthentication_attack
- Rogue Access Points: https://en.wikipedia.org/wiki/Rogue_access_point
- Evil twin networks: https://en.wikipedia.org/wiki/Evil_twin_(wireless_networks)
- Wireless Security in general: https://en.wikipedia.org/wiki/Wireless_security

CryptoCubik, 2019-08-01:
That doesn't even start to answer the question, dipshit.
Wifi deauth is easy to understand and do.
Rogue access points and evil twin networks are completely unrelated because the deauther has no traffic capture capabilities and doesn't actually act as an AP it seems to just be creating fake networks (if it did why isn't it advertised, like, at all? anywhere at all?).
And that last link is just insulting.

Sorry for calling you a dipshit, but your response is offensive.
I suggest you think before posting next time.

projectb-temp, 2019-08-01:
Then maybe you should provide more information about what you've done so far and what you've set your system up to do? Just stating that you don't know what all the settings do is no excuse, if there's a well written Wiki for the project.

Maybe it creates 60 SSIDs because it is set to the value of `60` in the settings? We don't know, as we don't know your settings of SSIDs entered in the SSIDs section.

closed 2019-08-11 by CryptoCubik