SpacehuhnTech/esp8266_deauther #1136: The web administration page cannot be cleared.

zhangkai9494, 2019-08-12:
Hey buddy, I have a problem. After I remove your firmware and install the original firmware, I visit the web interface for management, but the page is still the firmware. What should I do?

projectb-temp, 2019-08-12:
Hi there.
Provide more information, please.
How did you "remove" the firmware and what do you mean by "original firmware"?
What device are you using?
Have you tried this? => https://github.com/spacehuhn/esp8266_deauther/tree/master/Reset_Sketch If so: What was the result?

zhangkai9494, 2019-08-12:
I use the new firmware to cover your firmware, but the web management page is still the firmware you provided, and the original function has been lost. The 4MB module I used, now what should I do if I want to completely remove your firmware? I just used this chip, thank you for your help.
    NodeMCU ESP8266

projectb-temp, 2019-08-12:
Please see my previous post. Everything relevant to blank the device is written there.

zhangkai9494, 2019-08-15:
I have successfully solved this problem. Thank you very much.😂

closed 2019-08-15 by tobozo