SpacehuhnTech/esp8266_deauther #1138: Unable to connect to pwned network on iPad and Mac

VincentLeChef, 2019-08-16:
So I bought a USB WiFi deauther from maltronics, i plugged it in my mac, when i connected to the network on my mac, it automatically disconnected me from the network. I tried conencting to the network on my Ipad and when I put the password "deauther" it kept saying incorrect password.

VincentLeChef, 2019-08-16:
now that im trying to connect to it again, there is no pwned network at all

projectb-temp, 2019-08-21:
Did you tried to reset the device to defaults or maybe tried to re-flash a new .bin file?
What about other (non Apple) devices you could try to connect to the AP - do they work?

VincentLeChef, 2019-08-21:
how can i reset the device? I have tried connecting it to my android phone but once it's connected it automatically disconects

projectb-temp, 2019-08-22:
Please see the project's Wiki for this: https://github.com/spacehuhn/esp8266_deauther/wiki/FAQ#how-to-do-a-reset

closed 2019-10-09 by tobozo