SpacehuhnTech/esp8266_deauther #1140: BOZO--Trying to remove your board from Arduino IDE in boards manager

tytower, 2019-08-18:
BOZO--Trying to remove your board from Arduino IDE in boards manager gives a null pointer exception. 

Please explain why I can't remove your board contribution  when I want to . Also please revisit the previous issue  and answer my question . Labels are not an answer!

```Warning: file /root/.arduino15/packages/esp8266/tools/python/3.7.2-post1/python links to an absolute path /usr/bin/python

java.lang.NullPointerException
java.lang.RuntimeException: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onRemovePressed$3(ContributionManagerUI.java:213)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at processing.app.BaseNoGui.onBoardOrPortChange(BaseNoGui.java:679)
	at processing.app.Base.onBoardOrPortChange(Base.java:1317)
	at processing.app.Base$12.actionPerformed(Base.java:1544)
	at processing.app.Base.rebuildBoardsMenu(Base.java:1494)
	at processing.app.Base$9.onIndexesUpdated(Base.java:1382)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onRemovePressed$3(ContributionManagerUI.java:211)
	... 1 more
java.lang.NullPointerException
java.lang.RuntimeException: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onRemovePressed$3(ContributionManagerUI.java:213)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributionInstaller.findAndExecutePreUninstallScriptIfAny(ContributionInstaller.java:197)
	at cc.arduino.contributions.packages.ContributionInstaller.remove(ContributionInstaller.java:243)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onRemovePressed$3(ContributionManagerUI.java:210)
	... 1 more
java.lang.NullPointerException
java.lang.RuntimeException: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onRemovePressed$3(ContributionManagerUI.java:213)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributionInstaller.findAndExecutePreUninstallScriptIfAny(ContributionInstaller.java:197)
	at cc.arduino.contributions.packages.ContributionInstaller.remove(ContributionInstaller.java:243)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onRemovePressed$3(ContributionManagerUI.java:210)
	... 1 more
java.lang.NullPointerException
	at processing.app.BaseNoGui.onBoardOrPortChange(BaseNoGui.java:679)
	at processing.app.Base.onBoardOrPortChange(Base.java:1317)
	at processing.app.Base$12.actionPerformed(Base.java:1544)
	at processing.app.Base.rebuildBoardsMenu(Base.java:1494)
	at processing.app.Base.openBoardsManager(Base.java:1400)
	at processing.app.Base$10.actionPerformed(Base.java:1422)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:842)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:886)
	at java.awt.Component.processMouseEvent(Component.java:6539)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6304)
	at java.awt.Container.processEvent(Container.java:2239)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
	at java.awt.Container.dispatchEventImpl(Container.java:2283)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

<details>
  <summary> Compile log (...)</summary>

  

</details>
<br><br><br>

tytower, 2019-08-18:
[image]

tobozo, 2019-08-18:
Although I know a bit of Java, I try to stay as far away as I can from those pesky stack traces. 
I remember reading one of those far back in 2007 and only managed to recover bits of my own sanity, still recovering.

Moreover, I believe Java was created to submit the human kind into permanent suffering while turning the computers into ovens. A new kind of himmel that is.

The laws of thermodynamics even allow us to suggest that Java could be the established cause of global warming!

Now while I'm not trying to put the climate change on you, please don't try to put your broken Arduino IDE on me.

Breaking things is a valid approach to learn new things, so if you know what you've broken here, it's only a matter of fixing it (e.g. by repairing or by breaking more things).

I'm trustful you'll eventually find the solution to this Arduino IDE problem.

Happy hacking !

spacehuhn, 2019-08-18:
[image]

tytower, 2019-08-18:
OK so it actually uninstalled as far as I can tell ,but did not get through to completion in the manager. Just left it hanging

closed 2019-08-19 by tytower