SpacehuhnTech/esp8266_deauther #1144: ERROR: Display not enabled with default .bin

HeinzAlberto, 2019-09-05:
Hi there!

I have flashed the default .bin of the 2.1.0-release and now I am not getting any display output at all...
I am using a normal I²C  SSD1306-OLED with a three-button config (menu up/down, enter).
The wires are all okay and the serial promt is working fine, too.

When I enter "screen on" in the serial promt, I get "ERROR: Display not enabled". 
What can I do against this problem?

Thank you very much!

unguest, 2020-05-10:
You can find many binaries that are matching your needs : https://github.com/SpacehuhnTech/esp8266_deauther/releases => some of them are made for oled-enabled devices (mostly based on NodeMCU & ATMega).

spacehuhn, 2020-06-16:
Maybe this helps https://github.com/SpacehuhnTech/esp8266_deauther/wiki/Setup-Display-&-Buttons

closed 2020-09-21 by stale[bot]